RFD-FHEM/SIGNALDuino #208: ESP32 Reset

romakrau, 2021-06-07:
Hallo zusammen,
ich finde in der src nicht die Stelle zum resetten des ESP32. Die einzige Stelle die ich gefunden habe ist in signalesp.h zeile 632. Dort wird ein esp_task_wdt_reset() aufgerufen obwohl ich keine Initialisierung des Watchdogs finden kann, müsste das nicht ESP.reset() lauten. Ferner sind Aufrufe yield() notwendig meiner Meinung nach ist das eine Funktion des 8266.
Gruss
Roman

sidey79, 2021-06-07:
Was genau willst Du denn erreichen?

romakrau, 2021-06-07:
Ich habe das Problem das der esp32 nach unregelmäßiger Zeit immer auf disconnect geht. Danach verbindet er sich zwar wieder aber Erkennung läuft nicht an. Ich muss ihn dann immer vom Strom trennen,  dann läuft er wieder. Ich wollte beim status disconnect halt ein notify mit einem Reset ausführen. 
Gruß Roman

HomeAutoUser, 2021-06-07:
Hallo @romakrau,
hast du eventuell mal bei dieser Erscheinung mit dem Terminal oder direkt auf die USB Schnittstelle gesehen was dort abläuft?
Es wäre Interessant zu erfahren ob es vielleicht Stacktrace Fehler ist oder ob das Programm einfach eine Fehlfunktion besitzt.

romakrau, 2021-06-08:
@HomeAutoUser : Dazu muss ich wohl den ESP32 mit Debug=on neu kompilieren und ihn an die USB-Schnittstelle hängen? Bisher habe ich nur das Logging in FHEM. Werde ich nachher mal versuchen.
Gruß
Roman

romakrau, 2021-06-08:
So, ich habe mal die Ausgabe angehangen. Jetzt läuft er gar nicht mehr. Was lief falsch?

08:54:32.253 -> 
08:54:32.253 -> Brownout detector was triggered
08:54:32.253 -> 
08:54:32.253 -> ets Jun  8 2016 00:22:57
08:54:32.253 -> 
08:54:32.253 -> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
08:54:32.301 -> configsip: 0, SPIWP:0xee
08:54:32.301 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
08:54:32.301 -> mode:DIO, clock div:1
08:54:32.301 -> load:0x3fff0018,len:4
08:54:32.301 -> load:0x3fff001c,len:1216
08:54:32.301 -> ho 0 tail 12 room 4
08:54:32.301 -> load:0x40078000,len:10944
08:54:32.301 -> load:0x40080400,len:6388
08:54:32.301 -> entry 0x400806b4
08:54:32.655 -> 
08:54:32.655 -> Brownout detector was triggered
08:54:32.655 -> 
08:54:32.655 -> ets Jun  8 2016 00:22:57
08:54:32.655 -> 
08:54:32.655 -> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
08:54:32.655 -> configsip: 0, SPIWP:0xee
08:54:32.655 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
08:54:32.655 -> mode:DIO, clock div:1
08:54:32.655 -> load:0x3fff0018,len:4
08:54:32.655 -> load:0x3fff001c,len:1216
08:54:32.655 -> ho 0 tail 12 room 4
08:54:32.655 -> load:0x40078000,len:10944
08:54:32.655 -> load:0x40080400,len:6388
08:54:32.655 -> entry 0x400806b4
08:54:33.003 -> 
08:54:33.003 -> Brownout detector was triggered
08:54:33.003 ->

HomeAutoUser, 2021-06-08:
Wie war deine Vorgehensweise?

romakrau, 2021-06-08:
Kompilieren mit Arduino 1.8.12 Version 
version | V 3.5.0-dev+20201221 SIGNALESP cc1101 (chip CC1101) DBG - compiled at Jun  8 2021 10:51:22

fhem Module und Protokol 
versionProtocols | 1.30
versionmodul | 3.5.1+20210525

Anschluß an USB und Hochladen per Arduino. Eingbunden in FHEM:
Internals:
   Clients    :CUL_EM:CUL_FHTTK:CUL_TCM97001:CUL_TX:CUL_WS:Dooya:FHT:FLAMINGO:FS10:FS20: :Fernotron:Hideki:IT:KOPP_FC:LaCrosse:OREGON:PCA301:RFXX10REC:Revolt:SD_AS: :SD_BELL:SD_GT:SD_Keeloq:SD_RSL:SD_UT:SD_WS07:SD_WS09:SD_WS:SD_WS_Maverick:SOMFY: :Siro:SIGNALduino_un:
   DEF        192.168.187.244:23
   DMSG       nothing
   DevState   disconnected
   DeviceName 192.168.187.244:23
   FUUID      60994941-f33f-e93f-c477-f15d56304b1fc468
   LASTDMSG   nothing
   LASTDMSGID nothing
   NAME       SGD1
   NEXT_OPEN  1623164244.39261
   NR         276
   PARTIAL    
   STATE      disconnected
   TIME       1622739090
   TYPE       SIGNALduino
   cc1101_available 1

Gelogt in FHEM:
2021-06-08_10:59:24 SGD1 ping: OK
2021-06-08_11:01:24 SGD1 ping: OK
2021-06-08_11:03:24 SGD1 ping: OK
2021-06-08_11:06:24 SGD1 ping: OK
2021-06-08_11:08:24 SGD1 ping: OK
2021-06-08_11:10:24 SGD1 DISCONNECTED
2021-06-08_11:10:24 SGD1 CONNECTED
2021-06-08_11:10:26 SGD1 opened
2021-06-08_11:10:27 SGD1 cc1101_config: Freq: 868.300 MHz, Bandwidth: 406 KHz, rAmpl: 42 dB, sens: 4 dB, DataRate: 17257.69 Baud
2021-06-08_11:10:27 SGD1 cc1101_config_ext: Modulation: 2-FSK, Syncmod: 16/16 + carrier-sense above threshold
2021-06-08_11:10:28 SGD1 cc1101_patable: C3E = 00 84 00 00 00 00 00 00 => 5_dBm
2021-06-08_11:12:26 SGD1 ping: OK
2021-06-08_11:14:26 SGD1 ping: OK
2021-06-08_11:16:26 SGD1 ping: OK
2021-06-08_11:18:26 SGD1 ping: OK
2021-06-08_11:20:26 SGD1 DISCONNECTED
2021-06-08_11:20:27 SGD1 CONNECTED
2021-06-08_11:20:29 SGD1 opened
2021-06-08_11:20:30 SGD1 cc1101_config: Freq: 868.300 MHz, Bandwidth: 406 KHz, rAmpl: 42 dB, sens: 4 dB, DataRate: 17257.69 Baud
2021-06-08_11:20:30 SGD1 cc1101_config_ext: Modulation: 2-FSK, Syncmod: 16/16 + carrier-sense above threshold
2021-06-08_11:20:30 SGD1 cc1101_patable: C3E = 00 84 00 00 00 00 00 00 => 5_dBm
2021-06-08_11:22:29 SGD1 ping: OK
2021-06-08_11:24:29 SGD1 ping: OK
2021-06-08_11:26:29 SGD1 ping: OK
2021-06-08_11:28:29 SGD1 ping: OK
2021-06-08_11:30:29 SGD1 ping: OK
2021-06-08_11:32:29 SGD1 ping: OK
2021-06-08_11:34:29 SGD1 DISCONNECTED
2021-06-08_11:34:29 SGD1 CONNECTED
2021-06-08_11:34:31 SGD1 opened
2021-06-08_11:34:32 SGD1 cc1101_config: Freq: 868.300 MHz, Bandwidth: 406 KHz, rAmpl: 42 dB, sens: 4 dB, DataRate: 17257.69 Baud
2021-06-08_11:34:32 SGD1 cc1101_config_ext: Modulation: 2-FSK, Syncmod: 16/16 + carrier-sense above threshold
2021-06-08_11:34:33 SGD1 cc1101_patable: C3E = 00 84 00 00 00 00 00 00 => 5_dBm
2021-06-08_11:36:31 SGD1 ping: OK
2021-06-08_11:38:31 SGD1 ping: OK
2021-06-08_11:40:31 SGD1 ping: OK
2021-06-08_11:42:31 SGD1 ping: OK
2021-06-08_11:44:31 SGD1 ping: OK
2021-06-08_11:46:31 SGD1 ping: OK
2021-06-08_11:48:31 SGD1 ping: OK
2021-06-08_11:50:31 SGD1 ping: OK
2021-06-08_11:52:31 SGD1 ping: OK
2021-06-08_11:54:31 SGD1 ping: OK
2021-06-08_11:56:31 SGD1 ping: OK
2021-06-08_11:58:31 SGD1 ping: OK
2021-06-08_12:00:31 SGD1 ping: OK
2021-06-08_12:02:31 SGD1 ping: OK
2021-06-08_12:04:31 SGD1 ping: OK
2021-06-08_12:06:31 SGD1 ping: OK
2021-06-08_12:08:31 SGD1 ping: OK
2021-06-08_12:10:31 SGD1 ping: OK
2021-06-08_12:15:32 SGD1 DISCONNECTED

Output fhem.log:

2021.06.08 11:34:29 1: 192.168.187.244:23 disconnected, waiting to reappear (SGD1)
2021.06.08 11:34:29 1: SGD1: DoInit, 192.168.187.244:23
2021.06.08 11:34:29 1: 192.168.187.244:23 reappeared (SGD1)
2021.06.08 11:34:31 3: SGD1: SimpleWrite_XQ, disable receiver (XQ)
2021.06.08 11:34:31 3: SGD1: StartInit, get version, retry = 0
2021.06.08 11:34:31 2: SGD1: CheckVersionResp, initialized 3.5.1+20210525
2021.06.08 11:34:31 3: SGD1: CheckVersionResp, enable receiver (XE) 
2021.06.08 11:34:33 3: SGD1: CheckCcpatableResponse, patable: 84
2021.06.08 12:13:31 3: SGD1: KeepAlive, not ok, retry = 2 -> get ping
2021.06.08 12:14:31 3: SGD1: KeepAlive, not ok, retry = 3 -> get ping
2021.06.08 12:15:31 3: SGD1: KeepAlive, not ok, retry count reached. Reset
2021.06.08 12:15:31 3: SGD1: ResetDevice, ESP32
2021.06.08 12:15:31 3: Opening SGD1 device 192.168.187.244:23
2021.06.08 12:15:32 1: SGD1: Can't connect to 192.168.187.244:23: 192.168.187.244: No route to host (113)
2021.06.08 12:15:32 3: SGD1: Connect, 192.168.187.244: No route to host (113)

Eine Verbindung via Serieller Schnittstelle scheitert.

HomeAutoUser, 2021-06-08:
@romakrau
1) Was möchtest du überhaupt empfangen?
Ich habe einen ESP32 problemlos flashen können und in FHEM kommen auch Daten. (getestet mit OOK / FSK || FW 3.5.0-dev+20210420)
2) Welche Hardware hast du genau?

`2021-06-08_12:10:31 SGD1 ping: OK`
Deutet darauf hin, das keine Daten kommen und SIGNALduino ein Ping setzt.
Wenn keine Daten kommen, so geht auch die Hardware meines Wissens auf Closed.
`2021-06-08_12:15:32 SGD1 DISCONNECTED`

romakrau, 2021-06-09:
Ich möchte gerne Werte von einem TX37 Sensor empfangen und in FHEM verarbeiten. Ich benutze einen ESP32 Devkit V1.

Es ist nicht einfach da eine vernünftige Debug-Ausgabe zu bekommen. Anbei mal eine Ausgabe:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

Starting config portal with SSID: NodeDuinoConfig
*WM: [1] AutoConnect
*WM: [2] ESP32 event handler enabled
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
*WM: [1] Connecting to SAVED AP: XXXXXXXXXXXXXXXXXXX
*WM: [3] Using Password: XXXXXXXXXXXXXXXXx
*WM: [3] WiFi station enable
*WM: [1] connectTimeout not set, ESP waitForConnectResult...
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS
*WM: [1] STA IP Address: 192.168.187.244
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started
*WM: [2] WiFi Scan ASYNC completed in 5105 ms
*WM: [2] WiFi Scan ASYNC found: 5
*WM: [2] WiFi Scan completed in 5106 ms
WiFi lost connection. Reason: 1735289158
*WM: [2] [EVENT] WIFI_REASON: 200

Deiner Beschreibung nach müsste z.B. bei einem Bewegungsmelder ohne Detection die Hardware abschalten?
Dazu kommt der Ping und das Disconnect nicht periodisch genug z.B. nach 20-mal Ping etc...

Vielleicht kann ja jemand mit dem Output was anpfangen?

Gruß und Danke Roman

HomeAutoUser, 2021-06-09:
Hallo Roman @romakrau,
hast du den TX37 bei dir auf einer Wetterstation im Blick? Es könnte sein, das du vielleicht Außerhalb der Reichweite bist. Ich habe bei meinem auch festgestellt, je nach baulicher Beschaffenheit, kann ein verrücken von einem halben Meter zum Abbruch führen.

Wieso das `DISCONNECTED` unterschiedlich auftritt, dazu kann dir vielleicht @sidey79 mehr sagen.

Welchen genauen Aufbau von deinem ESP32 besitzt du? Ich habe auch ein Devkit laufen und es bricht nicht ab. Welche Hardware hattest du beim kompilieren asgewählt oder hast du das File von Git genommen?

Mfg

romakrau, 2021-06-09:
Hallo ich habe die letzte Fassung mit Platformio kompiliert und aufgespielt. Den Sensor hatte ich daneben liegen. Der Abbruch kam vom Windowsmanager soweit ich das interpretiere. Das Devkit ist korrekt ausgewählt. Auch wenn die Arduino Umgebung manchmal spinnt. Leider bricht das Debugfenster (Mobaxterm) immer ab. Ich versuche mal die Stelle der Ausgabe im Sourcecode zu finden und dann ein ESP reset einzufügen. Was besseres fällt mir nicht ein. Gruss Roman

romakrau, 2021-06-10:
Ich habe jetzt die wifi-manager.h mal durchstöbert und sowohl den WM_RTC und esp32autoreconnect  gesetzt. Ich hoffe das ich dadurch weitere Informationen bezüglich des Disconnects erhalten. Melde mich sobald ich was neues habe.

romakrau, 2021-06-15:
Ich gebe auf. Der Fehler taucht immer wieder mal sporadisch auf. Ich vermute das Ganze in einer fehlerhaften Pulsauswertung. Die Belegung einer CONST während der Laufzeit ist sehr ungewöhnlich, ebenso die Definition einer Variable zur Laufzeit in einem IF Konstrukt.  Ferner ist diese Zeile falsch sDuration = -sDuration; , es muss lauten sDuration=-sDuration; . Ich nutze jetzt die Version von Ralf und bekomme korrekte Werte.
Gruß
Roman

HomeAutoUser, 2021-06-15:
Ich vermute, du suchst an der falschen Stelle. Es ist spanisch, das es bei anderer HW aber selbiger uC nicht zu dem Fehler kommt.

romakrau, 2021-06-15:
Ich finde es auch merkwürdig, allerdings sind die übermittelten Werte jetzt korrekt (Vergleich Wetterstation).  Ich habe schon einen neuen ESP32Devkit und CC1100 868MHz bestellt. Vielleicht liegt es ja an der Version. Diese kann übrigens nicht von dem Arduino Kompiler ausgelesen werden?!?

romakrau, 2021-07-05:
Nach meinem Urlaub habe ich jetzt ein neues ESP32 Devkit V1und CC1101 programmiert. Das Problem ist dasselbe. Nach einer zeit geht das modul immer in den Mode Disconnect und ist im Wlan nicht sichtbar. Vielleicht gibts ja irgendwann mal eine lösung.

sidey79, 2021-07-05:
Ich erinnere mich, für den esp8266 ein reconnect eingebaut zu haben. Finden konnte ich ihn auf Anhieb im code nicht mehr.

Vielleicht greift der nicht für den esp32.
Die verwendete Bibliothek wifimanger kann das zumindest auch. Vielleicht lohnt es sich hier das Update auf die neue Lib zu entwickeln und dann den reconnect zu aktivieren.

HomeAutoUser, 2021-07-05:
@romakrau auf welcher Frequenz läuft der cc1101? Hast du ggf. andere Sensoren auf einem anderen Frequenzbereich?

Ich denke das Problem liegt wo anders. Selbst ich suche gerade genau nach solch einem Phänomen bei einem weiteren Eigenbau. Grund dafür, das nur auf einem Frequenzbereich der uC in Disconnect geht. Der 2.uC auf einem anderen Band geht tatenlos. Sobald ich nun die Bänder tausche, so wandert der Fehler. Das habe ich reproduziert. Ich tippe ein spezielles Signal führt zum Crash. Problem, es dauert hier 24-96..h bis zum Auftreten.

@sidey79 reicht es aus, das Signal von der TTY mitzuzeichnen wenn ich den Fehler suche oder wie würdest du dich dort heran tasten?

sidey79, 2021-07-05:
Das ist quasi wie die Nadel im Heuhaufen suchen.
Dafür müsste man genau wissen, wann der ESP hängen bleibt.
Die Serielle Schnittstelle ist da schon ein guter Ansatz um zu schauen ob es einen Stacktrace gibt.
Der kann mit Hilfstools dann zumindest hinweise geben, wo im Code es zu Problemen kam.

romakrau, 2021-07-06:
Nein ein weiteres Band habe ich nicht. Ich interpretiere das mal so das du die Frequenz meinst. Der Sensor läuft auf 868MHz und die Daten sind korrekt. Ich habe jetzt eine Krücke eingebaut indem ich in der Loop folgendes prüfe:
 	if (WiFi.status() != WL_CONNECTED) {
		ESP.restart();
	}
Das läuft bisher schon seit 12Std.
Gruß
Roman

HomeAutoUser, 2021-07-08:
Hallo @romakrau 
besteht bei dir die Möglichkeit, das du den ESP nahe einem Laptop betreiben könntest?
Ich habe nun seit 3 Tagen zum Glück den Absturz und lausche nun über die TTY mit. Leider habe ich keine Ausgabe erhalten "außer die Signale selbst". Somit würde ich einen Stacktracefehler in meinem Falle ausschließen weil der uC auch nicht neu startet. Ich vermute er hängt in einer Schleife weil ich gezwungen bin den uC neu zu starten damit dieser wieder anläuft.
Nun habe ich neu gestartet und hoffe auf den erneuten Fall um zu sehen bei welchem Typ Nachrichten es auftritt.

Welche config hast du vom ESP32 vorgenommen? Kannst du mal bitte ein "config" posten?

@sidey79 ich habe die letzten RAW´s mal hier zusammengestellt und würde den Test wiederholen. Sollte es erneut auftreten so vergleiche ich und dann würde ich den selben Fall noch einmal mit der Datenreduktion aus vornehmen.

```
MC;LL=-1005;LH=939;SL=-525;SH=445;D=A8DA745ACA1E0BF75584664;C=485;L=90;R=248;
MC;LL=-1006;LH=939;SL=-519;SH=446;D=A8DA745AEA1E0BF75585716;C=484;L=91;R=245;
MC;LL=-1001;LH=943;SL=-515;SH=457;D=A8DA745ADA1E0BF75584EB4;C=485;L=90;R=248;
MC;LL=-1040;LH=904;SL=-555;SH=426;D=A8C2345AEC0C0B034AA65B4;C=487;L=90;R=214;
MC;LL=-1038;LH=905;SL=-556;SH=413;D=A8C2345ADC0;C=485;L=41;R=215;
MC;LL=-1058;LH=881;SL=-561;SH=402;D=181606954C;C=483;L=38;R=213;
MC;LL=-2697;LH=3126;SL=-1237;SH=1679;D=EF31FF21;C=1456;L=32;R=236;
Mu; à€;‘…†;²Ü„;³ ;„¸–;¥°š;†ÌŒ;·Š;DEgbb;C1;REC;
MC;LL=-986;LH=965;SL=-481;SH=467;D=A8C9345AC8A;C=483;L=43;R=224;
MC;LL=-977;LH=966;SL=-486;SH=474;D=E0B6755ACD;C=483;L=40;R=225;
MC;LL=-989;LH=950;SL=-495;SH=486;D=9345AE8BE0;C=486;L=38;R=223;
MC;LL=-985;LH=963;SL=-481;SH=505;D=9D56B73D;C=488;L=32;R=224;
MC;LL=-980;LH=970;SL=-479;SH=478;D=A8C9345AD8A;C=484;L=43;R=224;
MC;LL=-980;LH=970;SL=-479;SH=478;D=E0B6755AC556;C=484;L=47;R=224;
Ms;‘Ñ;²Ä;£·£;´Ú‡;d1!!!AAAAAA!!A!!AAA!AA!!A!A!AAAAAAA!!A;C1;S3;R24;O;m2;
Ms;‘Ö;²À;£»£;´ß‡;D;C1;S3;R24;O;m1;
Ms;‘Ð;²Ä;£¹£;´à‡;D;C1;S3;R24;O;m0;
Ms;‘ß;²±;£´£;´å‡;D;C1;S3;R24;O;
Ms;Ó;±Þ‡;²Ã;£¼£;D;C0;S3;R20;
MC;LL=-2701;LH=3142;SL=-1241;SH=1669;D=EFD1FFC1;C=1458;L=32;R=236;
MC;LL=-2702;LH=3126;SL=-1252;SH=1667;D=EFD1E;C=1457;L=19;R=236;
MC;LL=-994;LH=940;SL=-513;SH=456;D=A8F8B45AC8E60B8955F1DCC;C=483;L=90;R=39;
MC;LL=-1008;LH=937;SL=-510;SH=458;D=A8F8B45AE8E60B8955F0CBC;C=485;L=90;R=39;
MC;LL=-997;LH=951;SL=-502;SH=463;D=F168B5B1CC1712ABE2A3C;C=485;L=82;R=39;
MC;LL=-1017;LH=931;SL=-530;SH=436;D=A8F7345AC9DE0A677596F7E;C=485;L=91;R=22;
MC;LL=-1009;LH=944;SL=-535;SH=438;D=A8F7345AE9DE0A677597E0C;C=487;L=90;R=22;
MC;LL=-1009;LH=924;SL=-515;SH=453;D=A8F7345AD9DE0A6775967AC;C=483;L=90;R=23;
MC;LL=-1010;LH=914;SL=-540;SH=443;D=518468B598;C=484;L=40;R=214;
Mu;°ˆ;‘¢;¢¤‚;“ƒ;¤‘„;µ°€;d#!!!!A#!!!!!CA#!!!!!A#C!CCCP;C1;RD7;
Mu;°€‰;Øƒ;¢Åƒ;ƒ…;¤É‚;…í‚;¶Ú€;d!!!#EcEeeecCCEeeeecEcCCEeeec`;C5;RE5;
Mu;€”;¡Â‚;‚ô‚;³Þ€;„ðƒ;¥­ƒ;D#############EEEE########################EEEE########################EEEE#####;C0;RE4;O;
Mu; Î‚;‘ˆ‚;²Ö€;ƒ;„îƒ;¥¼ƒ;d!!#!!####TTTS#!!!#!!!!## ;C3;RE5;
Mu;°Ö€;ù‚;‚;£Å‚;„äƒ;¥Áƒ;D21114TTTR11221122112121114TTTR11221122112121114TTTR112211221;C2;RE5;O;
```

romakrau, 2021-07-08:
Asche auf mein Haupt, ich bekomme das Modul von sidey79 nicht mehr geladen. Versuche laut WIKI SIGNALduino Modul aktualisieren: update all https://raw.githubusercontent.com/RFD-FHEM/RFFHEM/master/controls_signalduino.txt aber das Modul bleibt die Version von RALF9. Sorry

sidey79, 2021-07-08:
Update force.
Ich würde es auch nicht als mein Modul bezeichnen, es ist ein Community Projekte.

romakrau, 2021-07-08:
Danke Sidey79, aber das Kind musste einen Namen haben. 
Hier mein List allerdings bekomme ich jetzt keine TX29 Daten.

Internals:
   Clients    :CUL_EM:CUL_FHTTK:CUL_TCM97001:CUL_TX:CUL_WS:Dooya:FHT:FLAMINGO:FS10:FS20: :Fernotron:Hideki:IT:KOPP_FC:LaCrosse:OREGON:PCA301:RFXX10REC:Revolt:SD_AS: :SD_BELL:SD_GT:SD_Keeloq:SD_RSL:SD_UT:SD_WS07:SD_WS09:SD_WS:SD_WS_Maverick:SOMFY: :Siro:SIGNALduino_un:
   DEF        192.168.187.247:23
   DMSG       nothing
   DevState   initialized
   DeviceName 192.168.187.247:23
   FD         24
   FUUID      60e6fcfb-f33f-e93f-9472-00d76f45c3ba148e
   LASTDMSG   nothing
   LASTDMSGID nothing
   NAME       SGD2
   NR         283
   PARTIAL    
   STATE      opened
   TIME       1625752587
   TYPE       SIGNALduino
   cc1101_available 1
   sendworking 0
   version    V 3.5.0-dev+20210623 SIGNALESP cc1101 (chip CC1101) - compiled at Jul  8 2021 11:23:21
   versionProtocols 1.30
   versionmodul 3.5.2+20210616
   MatchList:
     10:SD_WS07 ^P7#[A-Fa-f0-9]{6}[AFaf][A-Fa-f0-9]{2,3}
     11:SD_WS09 ^P9#F[A-Fa-f0-9]+
     12:SD_WS   ^W\d+x{0,1}#.*
     13:RFXX10REC ^(20|29)[A-Fa-f0-9]+
     14:Dooya   ^P16#[A-Fa-f0-9]+
     15:SOMFY   ^Ys[0-9A-F]+
     16:SD_WS_Maverick ^P47#[A-Fa-f0-9]+
     17:SD_UT   ^P(?:14|20|24|26|29|30|34|46|56|68|69|76|78|81|83|86|90|91|91.1|92|93|95|97|99|104|105)#.*
     18:FLAMINGO ^P13\.?1?#[A-Fa-f0-9]+
     19:CUL_WS  ^K[A-Fa-f0-9]{5,}
     1:IT       ^i......
     20:Revolt  ^r[A-Fa-f0-9]{22}
     21:FS10    ^P61#[A-F0-9]+
     22:Siro    ^P72#[A-Fa-f0-9]+
     23:FHT     ^81..(04|09|0d)..(0909a001|83098301|c409c401)..
     24:FS20    ^81..(04|0c)..0101a001
     25:CUL_EM  ^E0.................
     26:Fernotron ^P82#.*
     27:SD_BELL ^P(?:15|32|41|42|57|79|96|98)#.*
     28:SD_Keeloq ^P(?:87|88)#.*
     29:SD_GT   ^P49#[A-Fa-f0-9]+
     2:CUL_TCM97001 ^s[A-Fa-f0-9]+
     30:LaCrosse ^(\S+\s+9 |OK\sWS\s)
     31:KOPP_FC ^kr\w{18,}
     32:PCA301  ^\S+\s+24
     3:SD_RSL   ^P1#[A-Fa-f0-9]{8}
     4:OREGON   ^(3[8-9A-F]|[4-6][0-9A-F]|7[0-8]).*
     5:CUL_TX   ^TX..........
     6:SD_AS    ^P2#[A-Fa-f0-9]{7,8}
     7:Hideki   ^P12#75[A-F0-9]+
     9:CUL_FHTTK ^T[A-F0-9]{8}
     X:SIGNALduino_un ^[u]\d+#.*
   QUEUE:
   READINGS:
     2021-07-08 15:58:05   cc1101_config   Freq: 868.350 MHz, Bandwidth: 325 KHz, rAmpl: 42 dB, sens: 4 dB, DataRate: 5603.79 Baud
     2021-07-08 15:58:05   cc1101_config_ext Modulation: ASK/OOK, Syncmod: No preamble/sync
     2021-07-08 15:57:17   cc1101_patable  C3E = 00 84 00 00 00 00 00 00 => 5_dBm
     2021-07-08 15:58:12   config          MS=1;MU=1;MC=1;Mred=1
     2021-07-08 15:59:15   ping            OK
     2021-07-08 15:57:15   state           opened
     2021-07-08 15:33:19   version         V 3.5.0-dev+20210623 SIGNALESP cc1101 (chip CC1101) - compiled at Jul  8 2021 11:23:21
   additionalSets:
   keepalive:
     ok         0
     retry      0
   mcIdList:
   mnIdList:
     100
   msIdList:
   muIdList:
Attributes:
   rfmode     Lacrosse_mode1
   whitelist_IDs 100

HomeAutoUser, 2021-07-08:
Klicke bitte auf attr rtmode und set. Dein Attribut hat den rfmode Wert aber dein cc110x zeigt die Konfiguration von ASK/OOK.

romakrau, 2021-07-08:
Hatte ich ge,acht, aber das dauerte eine Viertelstunde, war zwischendurch einkaufen ;-).

HomeAutoUser, 2021-07-08:
Dauert das einstellen ne viertel Stunde bei dir??

romakrau, 2021-07-08:
Ja scheint so, hatte aber nur nach dem Timestamp geschaut.
Hier zum Vergleich oben die Daten empfangen von Ralf9 und unten von Sidey79

2021.07.08 17:36:03 4: SGD1: Read, msg: MN;D=9206186A52AAAA00004866D8;R=10;
2021.07.08 17:36:03 4: SGD1: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2021.07.08 17:36:03 4: SGD2: Read, msg: MN;D=9206186A52AAAA00001220E8054E040600808462;R=19;
2021.07.08 17:36:03 4: SGD2: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)

romakrau, 2021-07-08:
2021.07.08 17:36:16 4: SGD1: Read, msg: MN;D=9206186A52AAAA00003E77DB;R=7;
2021.07.08 17:36:16 4: SGD1: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2021.07.08 17:36:16 4: SGD2: Read, msg: MN;D=9206;R=17;
2021.07.08 17:36:16 4: SGD2: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2021.07.08 17:36:16 4: SGD2: Parse_MN, Error! method ConvLaCrosse, Usage: Input #1, $hexData needs to be at least 8 chars long
2021.07.08 17:36:20 4: SGD1: Read, msg: MN;D=9206186A52AAAA000073DA31;R=8;
2021.07.08 17:36:20 4: SGD1: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2021.07.08 17:36:20 4: SGD2: Read, msg: MN;D=92;R=14;
2021.07.08 17:36:20 4: SGD2: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2021.07.08 17:36:20 4: SGD2: Parse_MN, Error! method ConvLaCrosse, Usage: Input #1, $hexData needs to be at least 8 chars long
2021.07.08 17:36:24 4: SGD1: Read, msg: MN;D=9206186A52AAAA0000034825;R=9;
2021.07.08 17:36:24 4: SGD1: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2021.07.08 17:36:24 4: SGD2: Read, msg: MN;D=9206186A52AAAA000040060414D00280F0808400D1040D0BC84980206B500C8105384134160100;R=18;
2021.07.08 17:36:24 4: SGD2: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2021.07.08 17:36:28 4: SGD1: Read, msg: MN;D=9206186A52AAAA000053F412;R=15;
2021.07.08 17:36:28 4: SGD1: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)
2021.07.08 17:36:28 4: SGD2: Read, msg: MN;D=9206186A52AAAA0000046479A494AC241800803111000430000007CD40441A88182E10;R=18;
2021.07.08 17:36:28 4: SGD2: Parse_MN, Found 2-FSK Protocol id 100 -> Lacrosse mode 1 with match (?^:^9)

elektron-bbs, 2021-07-08:
Mhmm, holt der ESP32 sooo langsam die Nachrichten vom CC1101 ab?
Normalerweise sollten die Nachrichten 8 Byte umfassen, so wie hier z.B.:

`MN;D=908652FD49012345;R=29;`

Oder hast du Register 0x03 FIFOTHR verstellt?

romakrau, 2021-07-08:
Hier die Register: 
Configuration register detail:
---------------------------------------------------------
add.  name       def.   cur.
0x00  IOCFG2   - 0x0D - 0x01
0x01  IOCFG1   - 0x2E - 0x2E
0x02  IOCFG0   - 0x2D - 0x46
0x03  FIFOTHR  - 0x47 - 0x01
0x04  SYNC1    - 0xD3 - 0x2D
0x05  SYNC0    - 0x91 - 0xD4
0x06  PKTLEN   - 0x3D - 0xFF
0x07  PKTCTRL1 - 0x04 - 0x00
0x08  PKTCTRL0 - 0x32 - 0x02
0x09  ADDR     - 0x00 - 0x00
0x0A  CHANNR   - 0x00 - 0x00
0x0B  FSCTRL1  - 0x06 - 0x06
0x0C  FSCTRL0  - 0x00 - 0x00
0x0D  FREQ2    - 0x10 - 0x21
0x0E  FREQ1    - 0xB0 - 0x65
0x0F  FREQ0    - 0x71 - 0xE8
0x10  MDMCFG4  - 0x57 - 0x59
0x11  MDMCFG3  - 0xC4 - 0x5C
0x12  MDMCFG2  - 0x30 - 0x06
0x13  MDMCFG1  - 0x23 - 0x22
0x14  MDMCFG0  - 0xB9 - 0xF8
0x15  DEVIATN  - 0x00 - 0x56
0x16  MCSM2    - 0x07 - 0x07
0x17  MCSM1    - 0x00 - 0x00
0x18  MCSM0    - 0x18 - 0x18
0x19  FOCCFG   - 0x14 - 0x16
0x1A  BSCFG    - 0x6C - 0x6C
0x1B  AGCCTRL2 - 0x07 - 0x07
0x1C  AGCCTRL1 - 0x00 - 0x68
0x1D  AGCCTRL0 - 0x91 - 0x90
0x1E  WOREVT1  - 0x87 - 0x87
0x1F  WOREVT0  - 0x6B - 0x6B
0x20  WORCTRL  - 0xF8 - 0xF8
0x21  FREND1   - 0xB6 - 0xB6
0x22  FREND0   - 0x11 - 0x11
0x23  FSCAL3   - 0xE9 - 0xEC
0x24  FSCAL2   - 0x2A - 0x2D
0x25  FSCAL1   - 0x00 - 0x14
0x26  FSCAL0   - 0x1F - 0x11
0x27  RCCTRL1  - 0x41 - 0x41
0x28  RCCTRL0  - 0x00 - 0x00
0x29  FSTEST   - N/A  - 0x59
0x2A  PTEST    - N/A  - 0x7F
0x2B  AGCTEST  - N/A  - 0x3F
0x2C  TEST2    - N/A  - 0x88
0x2D  TEST1    - N/A  - 0x31
0x2E  TEST0    - N/A  - 0x0B

elektron-bbs, 2021-07-08:
Register 0x03 passt...
Eigenartig, ich hatte hier mal eine andere Beobachtung aufgezeigt: https://github.com/RFD-FHEM/RFFHEM/pull/959#issuecomment-833634650
Da war der ESP8266 schneller als der Arduino.

romakrau, 2021-07-12:
Habe diese Zeile gefunden:
   if (FiFo.count()<120) yield();
Müsste das nicht <FIFO_LENGTH heissen? Was ist wenn der FIFO überläuft?

sidey79, 2021-07-12:
Nein, das passt so.

Wenn der FIFO vollständig befüllt ist, dann werden Daten verworfen.

romakrau, 2021-07-18:
So jetzt habe ich die Stelle gefunden an der der Serverabgeschaltet wird und der ESP32 nicht mehr sendet. Es ist die Routine  ethernetEvent() um die Zeile 600. Dort wird der serverClient gestoppt. Ich habe der Einfachheit halber mal ein ein ESP.restart() angehängt und der ESP läuft seit 5 Tagen ohne Probleme. Ebenso gibt es Wifi.onEvent Routine die den Server stoppen. Mir ist das Konzept noch nicht ganz klar geworden. Aber vielleicht hilft es ja das Problem zu lösen. 
Gruß
Roman

sidey79, 2021-07-18:
Hi Roman,

ich muss gestehen, ich habe in  ethernetEvent() keine Stelle gefunden, an welcher der Server gestoppt wird :-(

https://github.com/RFD-FHEM/SIGNALDuino/blob/89785b1a46dc8f2d51859140aa48714e48f0d004/src/signalesp.h#L567-L593

Vielleicht meinst Du serverClient.stop() ? Das dient dazu, eine Clientverbindung zu stoppen, wenn die Verbindung abgebochen ist. Abbrechen könnte diese z.B. durch WLAN Unterbrechung oder ähnliches geschehen.

romakrau, 2021-07-21:
Ich hatte ESP.restart() hinter die Zeile 590 gesetzt da ich den ESP zwar noch im Netz gesehen hatte, aber er keine Daten mehr empfangen hatte. Ich vermutete das der Client vielleicht noch gesetzt ist aber die Verbindung aus irgendeinem Grund weg war. Mit der Ergänzung von mir läuft sowohl Deine Software als auch die von Ralf seit dem 13ten problemlos.

romakrau, 2021-08-16:
Hallo zusammen,  dank es unermüdlichen Einsatz der Entwickler scheint das Problem, it der Version vom 08.08.2021 gelgelöst. War wohl eine EndlosEndlosschleife.  Gruss und Dank Roman

closed 2021-08-16 by romakrau