RFD-FHEM/SIGNALDuino #215: Senden mit CC1101 nicht möglich oder ich zu dumm?

Dattel, 2021-06-17:
Hi,
sagt mal, hat mal jemand versucht mit einem CC1101 Receiver/Sender per FHEM Nachrichten zu senden?
Ich versuche das jetzt schon zum X-ten mal und stelle mit irgendwie zu blöd an. 

Ich habe folgende Geräte für diesen Test am Start:
- ESP32 mit CC1101,  <-- dieser wird gerade zum Senden verwendet
- ESP8266 mit CC1101 <-- mit diesem habe ich es auch schon als Sender versucht
- ESP32 mit RXB8 

Ich habe bereits von beiden CC1101 versucht Nachrichten zu Senden, aber scheinbar kommt bei den andern beiden nichts an:
in FHEM habe ich mal exemplarisch VERBOSE auf 5 gesetzt, damit ich möglichst viel loggen kann. Auf allen Geräten ist Protokoll 3 auf der Whitelist - aber es kommt einfach nichts an.

```
021.06.17 12:19:25 3: SIGNALDuino32_dev: Attr, setting Verbose to: 5
2021.06.17 12:19:27 4: SIGNALDuino32_dev: KeepAlive, not ok, retry = 1 -> get ping
2021.06.17 12:19:27 5: SIGNALDuino32_dev: AddSendQueue, SIGNALDuino32_dev: P (1)
2021.06.17 12:19:27 4: SIGNALDuino32_dev: HandleWriteQueue, called
2021.06.17 12:19:27 4: SIGNALDuino32_dev: SendFromQueue, called
2021.06.17 12:19:27 5: SIGNALDuino32_dev: SimpleWrite, P
2021.06.17 12:19:27 4: SIGNALDuino32_dev: HandleWriteQueue, called
2021.06.17 12:19:27 4: SIGNALDuino32_dev: HandleWriteQueue, nothing to send, stopping timer
2021.06.17 12:19:34 5: SIGNALDuino32_dev: Set_sendMsg, msg=P3#00111010#R4
2021.06.17 12:19:34 5: SIGNALDuino32_dev: Set_sendMsg, IT V1 convertet tristate to bits=0000111111001100
2021.06.17 12:19:34 5: SIGNALDuino32_dev: Set_sendMsg, Preparing rawsend command for protocol=3, repeats=4, clock=250 bits=0000111111001100
2021.06.17 12:19:34 5: SIGNALDuino32_dev: AddSendQueue, SIGNALDuino32_dev: SR;R=4;P0=250;P1=-7750;P2=750;P3=-250;P4=-750;D=0104040404232323232323040423230404; (1)
2021.06.17 12:19:34 4: SIGNALDuino32_dev: Set_sendMsg, sending : SR;R=4;P0=250;P1=-7750;P2=750;P3=-250;P4=-750;D=0104040404232323232323040423230404;
2021.06.17 12:19:34 4: SIGNALDuino32_dev: HandleWriteQueue, called
2021.06.17 12:19:34 4: SIGNALDuino32_dev: SendFromQueue, called
2021.06.17 12:19:34 5: SIGNALDuino32_dev: SimpleWrite, SR;R=4;P0=250;P1=-7750;P2=750;P3=-250;P4=-750;D=0104040404232323232323040423230404;
2021.06.17 12:19:34 4: SIGNALDuino32_dev: SendFromQueue, msg=SR;R=4;P0=250;P1=-7750;P2=750;P3=-250;P4=-750;D=0104040404232323232323040423230404;
2021.06.17 12:19:36 4: SIGNALDuino32_dev: HandleWriteQueue, called
2021.06.17 12:19:36 4: SIGNALDuino32_dev: HandleWriteQueue, sendraw no answer (timeout)
2021.06.17 12:19:36 4: SIGNALDuino32_dev: HandleWriteQueue, nothing to send, stopping timer
```

Vielleicht kann mir jemand einen Tipp geben, wo ich ansetzen kann.

Sonnigen Gruß...

HomeAutoUser, 2021-06-25:
Hallo @Dattel
1) Welche Firmware nutzt du bei dem ESP´s?
2) Hast du mal deine Hardware überprüft?
3) Was möchtest du genau senden? / Sendest du von Hand oder von einem eingerichteten Device?
MfG

Dattel, 2021-06-25:
V 3.5.0-dev+20210420 SIGNALESP cc1101 (chip CC1101) - compiled at Jun 9 2021 20:58:57
Die Pin's habe ich mehrfach nach bestem Wissen/Gewissen kontrolliert und in der Config für mein Board angepasst.

Gesendet habe ich jeweils über FHEM sowohl mit SendMsg als auch RAW in unterschiedlichen Protokollen probiert
Beispiel: sendmsg -> P3#00111010#R4

Beim entsprechenden Empfänger waren jeweils diese Protokolle auf der WhiteList, aber mir scheint, als wenn das Senden schon nicht klappt... 

Wenn ich mir das Log anschaue, dann scheint es mir eher auf FHEM Seite schon zu klemmen, denn die DEBUG-Messages deuten darauf hin ([00_signalduino.pm Zeile 1689](https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/00_SIGNALduino.pm))  . Deshalb weiß ich nicht, ob mir hier Support geleistet wird oder ob ich damit vielleicht eher Richtung FHEM-Komponente mal die Frage stellen sollte.

HomeAutoUser, 2021-07-12:
Hallo @Dattel bist du weiter?
Also ich habe getestet und konnte keine Einschränkungen feststellen.

Hast du am cc1101 die LED angeschlossen um zu sehen ob der Ausgang "senden" aktiv ist?

Dattel, 2021-07-12:
hmm... ne ich habe selber seit dem noch nicht weiter probiert :-/
das mit der LED ist mal ne gute Idee... das werde ich bei Gelegenheit mal testen...