RFD-FHEM/SIGNALDuino #221: Hardware crashed after "certain unknown" signal or specific occurrence

HomeAutoUser, 2021-07-12:
... Sammlung der Fakten ...

Hardware: Arduino Pro Mini 3.3V 8Mhz
Firmware: V 3.5.0-dev+20210623 SIGNALduino cc1101

Der Fehler ist "reproduzierbar" indem das Unbekannte Signal erneut auftritt.
Sogar wenn die Hardware getauscht wurde, so tritt der Fehler auch auf oder wandert zu dem bestimmten Frequenzband.
Es handelt sich hierbei um ein Signal was auf dem OOK/ASK Frequenzband 433Mhz sendet.

Der uC bringt kein Stacktrace oder andere Meldung via TTY. Um den Normalzustand wieder herbeizuführen, muss der uC resettet /neu gestartet werden.

Anbei schonmal stets die letzten Signale vor dem Absturz.

1) - mit Komprimierung
```
Mu; à€;‘…†;²Ü„;³ ;„¸–;¥°š;†ÌŒ;·Š;DEgbb;C1;REC;
MC;LL=-986;LH=965;SL=-481;SH=467;D=A8C9345AC8A;C=483;L=43;R=224;
MC;LL=-977;LH=966;SL=-486;SH=474;D=E0B6755ACD;C=483;L=40;R=225;
MC;LL=-989;LH=950;SL=-495;SH=486;D=9345AE8BE0;C=486;L=38;R=223;
MC;LL=-985;LH=963;SL=-481;SH=505;D=9D56B73D;C=488;L=32;R=224;
MC;LL=-980;LH=970;SL=-479;SH=478;D=A8C9345AD8A;C=484;L=43;R=224;
MC;LL=-980;LH=970;SL=-479;SH=478;D=E0B6755AC556;C=484;L=47;R=224;
Ms;‘Ñ;²Ä;£·£;´Ú‡;d1!!!AAAAAA!!A!!AAA!AA!!A!A!AAAAAAA!!A;C1;S3;R24;O;m2;
Ms;‘Ö;²À;£»£;´ß‡;D;C1;S3;R24;O;m1;
Ms;‘Ð;²Ä;£¹£;´à‡;D;C1;S3;R24;O;m0;
Ms;‘ß;²±;£´£;´å‡;D;C1;S3;R24;O;
Ms;Ó;±Þ‡;²Ã;£¼£;D;C0;S3;R20;
MC;LL=-2701;LH=3142;SL=-1241;SH=1669;D=EFD1FFC1;C=1458;L=32;R=236;
MC;LL=-2702;LH=3126;SL=-1252;SH=1667;D=EFD1E;C=1457;L=19;R=236;
MC;LL=-994;LH=940;SL=-513;SH=456;D=A8F8B45AC8E60B8955F1DCC;C=483;L=90;R=39;
MC;LL=-1008;LH=937;SL=-510;SH=458;D=A8F8B45AE8E60B8955F0CBC;C=485;L=90;R=39;
MC;LL=-997;LH=951;SL=-502;SH=463;D=F168B5B1CC1712ABE2A3C;C=485;L=82;R=39;
MC;LL=-1017;LH=931;SL=-530;SH=436;D=A8F7345AC9DE0A677596F7E;C=485;L=91;R=22;
MC;LL=-1009;LH=944;SL=-535;SH=438;D=A8F7345AE9DE0A677597E0C;C=487;L=90;R=22;
MC;LL=-1009;LH=924;SL=-515;SH=453;D=A8F7345AD9DE0A6775967AC;C=483;L=90;R=23;
MC;LL=-1010;LH=914;SL=-540;SH=443;D=518468B598;C=484;L=40;R=214;
Mu;°ˆ;‘¢;¢¤‚;“ƒ;¤‘„;µ°€;d#!!!!A#!!!!!CA#!!!!!A#C!CCCP;C1;RD7;
Mu;°€‰;Øƒ;¢Åƒ;ƒ…;¤É‚;…í‚;¶Ú€;d!!!#EcEeeecCCEeeeecEcCCEeeec`;C5;RE5;
Mu;€”;¡Â‚;‚ô‚;³Þ€;„ðƒ;¥­ƒ;D#############EEEE########################EEEE########################EEEE#####;C0;RE4;O;
Mu; Î‚;‘ˆ‚;²Ö€;ƒ;„îƒ;¥¼ƒ;d!!#!!####TTTS#!!!#!!!!## ;C3;RE5;
Mu;°Ö€;ù‚;‚;£Å‚;„äƒ;¥Áƒ;D21114TTTR11221122112121114TTTR11221122112121114TTTR112211221;C2;RE5;O;

```
2) - mit Komprimierung
```
Mu;°ã€;‘ú€;¢Õ‚;ƒì‚;„Ûƒ;¥Îƒ;D#$TTTQ##!#!!!#!!###!#!##$TTTQ##!#!!!#!!###!#!##$TTTQ##!#!!!#!!###;C1;REB;O;
Mu;°ì€;ì‚;’ô€;£ß‚;„Öƒ;¥Îƒ;D212114TTTR1121222122111212114TTTR1121222122111212114TTTR11212221221;C2;REB;O;
Ms;‘×;²½;¤¶£;µß‡;dA!!!QQQQQQ!!Q!!QQQ!QQ!Q!!Q!QQQQQQQQQ!;C1;S4;R24;O;m2;
Ms;‘Ü;²±;¤»£;µâ‡;D;C1;S4;R24;O;m1;
Ms;‘Ð;²Ç;¤½£;µÙ‡;D;C1;S4;R24;O;m0;
Ms;‘Ú;²¼;¤¾£;µÕ‡;D;C1;S4;R24;O;
Ms;Ü;±Ý‡;²»;£¸£;D;C0;S3;R21;
MC;LL=-1003;LH=943;SL=-510;SH=458;D=A8C2345AC9BC0B1D4AFBC0C;C=485;L=90;R=232;
MC;LL=-995;LH=951;SL=-496;SH=480;D=A8C2345AE9BC0B1D4AFAD7C;C=486;L=90;R=232;
MC;LL=-997;LH=942;SL=-491;SH=467;D=A8C2345AD9BC0B1D4AFB4DC;C=482;L=90;R=232;
MC;LL=-1006;LH=939;SL=-519;SH=453;D=A8C9345ACDC60A995584CE4;C=486;L=90;R=247;
MC;LL=-1006;LH=944;SL=-508;SH=450;D=A8C9345AEDC60A995585D96;C=484;L=91;R=247;
MC;LL=-995;LH=960;SL=-506;SH=454;D=A8C9345ADDC60A995584434;C=485;L=90;R=247;
MC;LL=-996;LH=936;SL=-523;SH=450;D=A8F8B45ACA460BF755C45BC;C=484;L=90;R=42;
MC;LL=-998;LH=944;SL=-508;SH=462;D=8F8B45AEA460BF755C54CC;C=485;L=86;R=41;
MC;LL=-1016;LH=929;SL=-520;SH=447;D=A8F8B45ADA460BF755C4D6E;C=485;L=91;R=41;
MC;LL=-2680;LH=3158;SL=-1235;SH=1681;D=E33EFF22;C=1458;L=32;R=234;
MC;LL=-2606;LH=3133;SL=-1216;SH=1788;D=E33EFF22;C=1457;L=32;R=234;
MC;LL=-1015;LH=935;SL=-525;SH=444;D=A8F7345AC8BE0A9955AF764;C=486;L=90;R=29;
MC;LL=-1001;LH=934;SL=-511;SH=458;D=CD16BA2F82A6556B9858;C=483;L=77;R=29;
MC;LL=-1015;LH=920;SL=-532;SH=443;D=A8F7345AD8BE0A9955AFFB4;C=484;L=90;R=29;
MC;LL=-955;LH=986;SL=-491;SH=481;D=A8DA745AC80;C=485;L=41;R=225;
MC;LL=-960;LH=964;SL=-483;SH=482;D=182FDD5680;C=481;L=37;R=225;
MC;LL=-955;LH=964;SL=-464;SH=509;D=51B4E8B5D0;C=481;L=40;R=224;
MC;LL=-977;LH=986;SL=-482;SH=475;D=C17EEAB418;C=486;L=38;R=225;
MC;LL=-973;LH=978;SL=-483;SH=482;D=A8DA745AD80;C=485;L=41;R=225;
MC;LL=-938;LH=980;SL=-493;SH=481;D=8C17EEAB40;C=481;L=38;R=225;
Ms;‘Ó;²¿;¤¿£;µå‡;dA!!!QQQQQQ!!Q!!QQQ!QQ!Q!!Q!QQQQQQQQQ!;C1;S4;R24;O;m2;
Ms;‘Ï;²È;¤¿£;µÛ‡;D;C1;S4;R24;O;m1;
Ms;‘Ò;²Ä;¤¿£;µÜ‡;D;C1;S4;R24;O;m0;
Ms;‘Ô;²Á;¤Ã£;µß‡;D;C1;S4;R24;O;
Ms;ß;±á‡;²°;£Ä£;D;C0;S3;R23;
```

3) - ohne Komprimierung

... folgt noch

HomeAutoUser, 2021-07-15:
@sidey79 & @elektron-bbs, ich denke nachgewiesen zu haben, das der uC sich nicht aufhängt sondern in einer Schleife verharrt.
Mit DEBUG Ausgaben sehe ich eine fortlaufende Schleife welche das LOG füllt und nicht zu normal zurück kehrt.

```
þfž˜€†xæà†ž—ŠÒ’Mé)_p0s0l1H(vcnt:2:RES:)_p0l1L0s0l1L0H(vcnt:5:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_l1H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0p0l1L0H(vcnt:3:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1L0s0l1L0H(vcnt:5:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_l1H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0p0l1L0H(vcnt:3:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1L0s0l1L0H(vcnt:5:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_l1H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0p0l1L0H(vcnt:3:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1L0s0l1L0H(vcnt:5:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_l1H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0p0l1L0H(vcnt:3:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1L0s0l1L0H(vcnt:5:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_l1H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0p0l1L0H(vcnt:3:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1L0s0l1L0H(vcnt:5:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RES:)_p0s0l1H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_l1H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0p0l1L0H(vcnt:3:RES:)_P1L0H(vcnt:1:RES:)_p0l1H(vcnt:1:RE [... cut ...] :)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0s0l1H(vcnt:7:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_l1H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0H(vcnt:0:RES:)_p0l1H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_P1L0H(vcnt:1:RES:)_p0l1L0H(vcnt:2:RES:)_p0l1L0s0l1L0l1H(vcnt:6:RES:)_p0l1L0H(
```

Nun bist du gefragt @sidey79 um auch dein Wissen einzubringen wo er hängt.
Den Grund wieso, konnte ich leider nicht festhalten mit DEBUG Ausgaben.

Ich denke wir nähern uns und müssen nun weiter schauen ins Innere.

sidey79, 2021-07-15:
Hast Du eine Idee, welche Nachricht zuvor empfangen wurde?

HomeAutoUser, 2021-07-15:
Leider habe ich genau den Ansatz nicht in der Aufzeichnung weil der Rechner neu startete :D Ich bleibe dran und versuche zu sammeln was an Erkenntnissen nützlich ist nach einer Woche Rechnerabstinenz. ;)

sidey79, 2021-07-25:
Ich habe so die Vermutung, dass es mit folgendem Abschnitt zu tun hat.

https://github.com/RFD-FHEM/SIGNALDuino/blob/4487c024aaa8b35e2a77b0237f53470fbce6222c/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L1520-L1536

HomeAutoUser, 2021-07-26:
@sidey79 nun bin ich wieder aktiv zum Teil.

Ich denke eher der Teil https://github.com/RFD-FHEM/SIGNALDuino/blob/4487c024aaa8b35e2a77b0237f53470fbce6222c/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L1543-L1646 wird es sein. Dieser läuft immer Endlos in dem Falle.

Ich habe nun im ersten Post (https://github.com/RFD-FHEM/SIGNALDuino/issues/221#issue-942409214) ganz unten den Fall ergänzt wo ich Glück hatte. Die Aufzeichnung lief als der Fehler auftrat und ich noch vorzeititg das Programm zum speichern brachte als der Rechner es "schloss wegen Speicherüberlauf" ;-)

```
�MC;LL=-1000;LH=944;SL=-523;SH=446;D=BAFBA06;C=485;L=27;R=215;�

 check:mcrst: MC found: 
mlen:254:mstart: 0
l1L0s0l1L0H(vcnt:4:RES:)_P1S1L0s0l1H(vc ....
```

hier ist der Übergang zu Schleife.

Hast du eine Idee weiter den Fall einzugrenzen?

sidey79, 2021-07-26:
> @sidey79 nun bin ich wieder aktiv zum Teil.
> 
> Ich denke eher der Teil
> 
> https://github.com/RFD-FHEM/SIGNALDuino/blob/4487c024aaa8b35e2a77b0237f53470fbce6222c/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L1543-L1646
> 
> wird es sein. Dieser läuft immer Endlos in dem Falle.

In diesen Abschnitt wird i immer erhöht und dadurch sehe ich keine Möglichkeit zu einer Endlosschleife.
Hast Du etwas konkretes in dem Codeabschnitt gefunden, was die Endlosschleife verursachen könnte?

> Ich habe nun im ersten Post ([#221 (comment)](https://github.com/RFD-FHEM/SIGNALDuino/issues/221#issue-942409214)) ganz unten den Fall ergänzt wo ich Glück hatte. Die Aufzeichnung lief als der Fehler auftrat und ich noch vorzeititg das Programm zum speichern brachte als der Rechner es "schloss wegen Speicherüberlauf" ;-)
> 
> ```
> �MC;LL=-1000;LH=944;SL=-523;SH=446;D=BAFBA06;C=485;L=27;R=215;�
> 
>  check:mcrst: MC found: 
> mlen:254:mstart: 0
> l1L0s0l1L0H(vcnt:4:RES:)_P1S1L0s0l1H(vc ....
> ```
> 
> hier ist der Übergang zu Schleife.
> 
> Hast du eine Idee weiter den Fall einzugrenzen?

Ja, das müsste ich mal durch ein Debug Programm schicken. Dann wissen wir es.
Ich weiss jetzt aber nicht, ob ich das mit VSCode schon mal gemacht habe :)

HomeAutoUser, 2021-07-27:
> 
> Ja, das müsste ich mal durch ein Debug Programm schicken. Dann wissen wir es.
> Ich weiss jetzt aber nicht, ob ich das mit VSCode schon mal gemacht habe :)

Somit sollte ja deine Aufgabe klar sein ;-)
Es gibt Fälle, wo die Schleife weiter läuft. Die Tage werde ich Hilfsausgaben einfügen um noch genauer dies zu verdeutlichen.

sidey79, 2021-07-27:
Ja die Aufgabe ist klar ;)

Oder ich lade es in Visual Studio :)

HomeAutoUser, 2021-07-30:
@sidey79 ich liefere Dir nun den Beweis, das wir in der Schleife `while (i < pdec->messageLen)  {  }` hängen bleiben.

Bei dem Auftreten landen wir in der Endlosschleife welche sich immer wiederholt.  Ich habe beim Neustart der Schleife einen Zeilenumbruch hinzugefügt. Zusätzlich habe ich das i und  pdec messageLen ausgegeben am ENDE der Schleife (gekennzeichnet im Code unten). Desweiteren bin ich mir sicher, das der Fehler nur auftritt wenn  pdec messageLen = 254 ist.
```
PD sync: 481, -9012, TOL: 400, sFACT: -18.74, mstart: 0
 MS check: Index:  MStart: 0 SYNC: 3, CP: 2 - MEFound: 1
 - MEnd: 73
Filter Match: 
Ms;±¹;’á;£´£;´Ñ‡;D#!!!$$$$$$!!$!!$$$$!$$!$$!!$$$$$$$$$$;C2;S3;R19;O;m1;

PD sync: 456, -9014, TOL: 404, sFACT: -19.77, mstart: 0
 MS check: Index:  MStart: 0 SYNC: 3, CP: 2 - MEFound: 1
 - MEnd: 73
Filter Match: 
Ms;±Ï;’È;£¶£;´á‡;D#!!!$$$$$$!!$!!$$$$!$$!$$!!$$$$$$$$$$;C2;S3;R19;O;m0;

PD sync: 469, -9011, TOL: 404, sFACT: -19.21, mstart: 0
 MS check: Index:  MStart: 0 SYNC: 3, CP: 2 - MEFound: 1
 - MEnd: 73
Filter Match: 
Ms;±¹;’Õ;£³£;´â‡;D#!!!$$$$$$!!$!!$$$$!$$!$$!!$$$$$$$$$$;C2;S3;R19;O;

PD sync: 472, -9020, TOL: 36, sFACT: -19.11, mstart: 42
 MS check: Index:  MStart: 42 SYNC: 3, CP: 0 - MEFound: 1
 - MEnd: 115
Filter Match: 
Ms;Ø;±Ú‡;²º;£¼£;D;C0;S3;R17;
 check:mcrst: MU found: 
Mu; ý;‘ì€;¢é‚;ƒÜ‚;´õ€;…Íƒ;¦Øƒ;D!!#CCCA#CCA!#CCCCA#CA#A%eeea#A#CCCA!#A!!!#A#A!!#CCCA#CCA!#CCCCA#CA#A%eeea#A#CCCA!#A!!!#A#A!!#CCCA#CCA!#CCCCA#CA#A%eeea#A#CCCA!;C1;R9;O;
 check:mcrst: MU found: 
Mu; Ú‚;Ú‚;²÷€;“û€;„Öƒ;¥Õƒ;D###!!!#!!#!!!!#!###!!!#!!#!!!!#!#!!#!!#!!!!#!###!!!#!!#!!!!#!##TTTS#!!!####;C3;RB;O;
 check:mcrst: MC found: 
mlen:254:mstart: 0
l1H(vcnt:0:RES:)_while i2while pdec messageLen254
p0H(vcnt:0:RES:)_while i5while pdec messageLen254
P1L0H(vcnt:1:RES:)_while i9while pdec messageLen254
p0l1L0H(vcnt:2:RES:)_while i15while pdec messageLen254
P1L0s0l1H(vcnt:3:RES:)_while i22while pdec messageLen254
p0H(vcnt:0:RES:)_while i25while pdec messageLen254
P1L0H(vcnt:1:RES:)_while i29while pdec messageLen254
P1L0s0l1L0H(vcnt:4:RES:)_while i37while pdec messageLen254
p0l1L0s0l1H(vcnt:4:RES:)_while i46while pdec messageLen254
p0l1L0s0l1L0H(vcnt:5:RES:)_while i61while pdec messageLen254
P1L0s0l1H(vcnt:3:RES:)_while i68while pdec messageLen254
p0l1H(vcnt:1:RES:)_while i72while pdec messageLen254
p0s0l1L0s0l1H(vcnt:5:RES:)_while i82while pdec messageLen254
p0l1L0H(vcnt:2:RES:)_while i87while pdec messageLen254
P1L0H(vcnt:1:RES:)_while i91while pdec messageLen254
p0l1L0H(vcnt:2:RES:)_while i97while pdec messageLen254
P1L0s0l1H(vcnt:3:RES:)_while i104while pdec messageLen254
p0H(vcnt:0:RES:)_while i107while pdec messageLen254
P1L0H(vcnt:1:RES:)_while i111while pdec messageLen254
P1L0s0H(vcnt:2:RES:)_while i117while pdec messageLen254
P1L0H(vcnt:1:RES:)_while i121while pdec messageLen254
p0l1L0H(vcnt:2:RES:)_while i127while pdec messageLen254
p0l1L0s0l1H(vcnt:4:RES:)_while i136while pdec messageLen254
p0l1L0s0l1L0H(vcnt:5:RES:)_while i151while pdec messageLen254
P1L0s0l1H(vcnt:3:RES:)_while i158while pdec messageLen254
p0l1H(vcnt:1:RES:)_while i162while pdec messageLen254
p0s0l1L0s0l1H(vcnt:5:RES:)_while i172while pdec messageLen254
p0l1L0H(vcnt:2:RES:)_while i177while pdec messageLen254
P1L0H(vcnt:1:RES:)_while i181while pdec messageLen254
p0l1L0H(vcnt:2:RES:)_while i187while pdec messageLen254
P1L0s0l1H(vcnt:3:RES:)_while i194while pdec messageLen254
p0H(vcnt:0:RES:)_while i197while pdec messageLen254
P1L0H(vcnt:1:RES:)_while i201while pdec messageLen254
P1L0s0l1L0H(vcnt:4:RES:)_while i209while pdec messageLen254
p0l1L0s0l1H(vcnt:4:RES:)_while i218while pdec messageLen254
p0l1L0s0l1L0H(vcnt:5:RES:)_while i233while pdec messageLen254
P1L0H(vcnt:1:RES:)_while i237while pdec messageLen254
[... 201 lines cut ...]
        i++;
      }

      if (mc_sync) { // don't add bit if manchester processing was canceled
        ...
      }
      else {
        #ifdef DEBUGDECODE
          DBG_PRINT('_');

            ## !!!!!!!!! zusätzliche DEBUGausgaben HomeAutoUser !!!!!!!!!  ##
            #DBG_PRINT i#
            #DBG_PRINT while pdec messageLen#
            ## !!!!!!!!!  END !!!!!!!!!  ##

        #endif
      }
    } //    endif (mc_sync)
    i++;
  }
  ...
}
```

**_Wie kann ich Dir nun weiter helfen damit wir dies beheben?_**

sidey79, 2021-07-30:
Mir ist nicht zu helfen :)

Spaß beiseite.

Ich konnte den Debugger nicht mehr starten. Zuletzt hatte ich das mit VisualStudio gemacht, aber ich vermute durch die Änderung des Projektes auf VSCcode hat da jetzt was nicht mehr geklappt oder ich habe irgendwas vergessen.

Jetzt habe ich rumgefrickelt, besser kann ich es nicht ausdrücken und bin soweit, dass ich in VSCode debuggen kann :)
Eventuell brauche ich die MC Nachricht, welche das Problemchen verursacht oder halt die Daten aus dem Speicher.

sidey79, 2021-07-30:
Also den hier habe ich getestet:
`�MC;LL=-1000;LH=944;SL=-523;SH=446;D=BAFBA06;C=485;L=27;R=215;�`

Der führt zu keiner Endlosschleife.
Die Herausforderung ist auch, dass die MC Ausgabe erst kommt, nachdem die Verarbeitung fertig ist.
Ich bräuchte die Daten wenn er in der Endlosschleife steckt. Hast Du eine idee, wie wir das herausfinden könnten?
Vielleicht `wenn i=255` dann `print_out` oder sowas aufrufen

HomeAutoUser, 2021-07-31:
Das die Nachricht erst ausgegeben wird nachdem die Schleife durchlief habe ich auch schon bedacht. An die MC Nachricht kommen wir nicht heran, nur an den Pufferwert bzw. an die Inputsignalfolge.

Den print_out werde ich Anfang der Woche mal versuchen umzusetzen. Gibt es die Möglichkeit an die direkten Inputimpulse oder Zeiten heranzukommen welche dann weiter gegeben werden. Sozusagen die Daten bevor die Schleife startet?

Plan B könnte sein, die Schleife bei 255 und minlenght 254 abzubrechen um das Endlosverhalten (Überlauf) zu umgehen. Das stellt aber ein WorkAround letztendlich da.

sidey79, 2021-07-31:
https://github.com/RFD-FHEM/SIGNALDuino/blob/89785b1a46dc8f2d51859140aa48714e48f0d004/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L902

An der Stelle ausgeben, an der Du auch die messageLen ausgegeben hast wenn i=255

HomeAutoUser, 2021-08-02:
> 
> 
> https://github.com/RFD-FHEM/SIGNALDuino/blob/89785b1a46dc8f2d51859140aa48714e48f0d004/src/_micro-api/libraries/signalDecoder/src/signalDecoder.cpp#L902
> 
> An der Stelle ausgeben, an der Du auch die messageLen ausgegeben hast wenn i=255

Das ist wieder einfach gesagt aber geht nicht ohne weiteres an der Stelle. :-(

Weder
`SignalDetectorClass::printOut();` -> cannot call member function 'void SignalDetectorClass::printOut()' without object
oder
`printOut();` -> 'printOut' was not declared in this scope
funktionieren dort. Da muss mehr geändert werden :-(

sidey79, 2021-08-02:
Du musst nur das Objekt verwenden, dann geht es.

sidey79, 2021-08-02:
https://github.com/RFD-FHEM/SIGNALDuino/blob/89785b1a46dc8f2d51859140aa48714e48f0d004/src/_micro-api/libraries/signalDecoder/src/signalDecoder.h#L231

pdec->printOut(); sollte es tun

HomeAutoUser, 2021-08-02:
Ich habe starte nun den nächsten Anlauf um mehr Informationen dann zu erhalten :-)
Mal schauen wie lange es dauert.

HomeAutoUser, 2021-08-03:
@sidey79 ich bitte dich nun mit dem Debugger zu beschäftigen *smile*
Hier kommen die Daten von `pdec->printOut();` bei `i==255`
```
...
p0H(vcnt:0:RES:)_while i=255while pdec messageLen=254
printOut now:

Sync:  Clock: 231, Tol: 118, PattLen: 6 (6), Pulse: -240, 599, mStart: 254, MCD: 0, mtrunc: 0
Signal: 01023102310102310145454541023102323232310102310101010102323101010232323231023232310102310101023102310102310101010102323101010232323231023232310102310101023102310102310545454541023102323232310102310101010102323101010232323231023232310102310101023102310102.  [254]
Pattern:  P0: 6*[-623] P1: 6*[231] P2: 3*[599] P3: 3*[-243] P4: 0*[-839] P5: 0*[857]
l1H(vcnt:0:RES:)_while i=2while pdec messageLen=254
...
p0H(vcnt:0:RES:)_while i=255while pdec messageLen=254
printOut now:

Sync:  Clock: 231, Tol: 118, PattLen: 6 (6), Pulse: -240, 599, mStart: 254, MCD: 0, mtrunc: 0
Signal: 01023102310102310145454541023102323232310102310101010102323101010232323231023232310102310101023102310102310101010102323101010232323231023232310102310101023102310102310545454541023102323232310102310101010102323101010232323231023232310102310101023102310102.  [254]
Pattern:  P0: 6*[-623] P1: 6*[231] P2: 3*[599] P3: 3*[-243] P4: 0*[-839] P5: 0*[857]
l1H(vcnt:0:RES:)_while i=2while pdec messageLen=254
...
```

Ich werde das Ganze noch einmal neu starten und beim 2. Auftreten dann analog die Daten hier veröffentlichen.

sidey79, 2021-08-03:
Ist der uC denn mit einer dieser Nachrichten hängen geblieben?

sidey79, 2021-08-03:
okay ich habe den Fehler grundsätzlich gefunden. 
Die Behebung muss ich mir erst noch mal ansehen

In Zeile 1662 hilft zumindest das Anpassen dieser Zeile, die Nebenwirkung davon schaue ich mir morgen an.

`		if (i<255) i++;`

HomeAutoUser, 2021-08-04:
Hallo @sidey79, hier sind noch die Daten aus dem weiteren Fehlverhalten. (Endlosschleife) Somit hast du 2 Datensätze wo die Schleife dauerhaft läuft.
```
...
p0H(vcnt:0:RES:)_while i=255while pdec messageLen=254
printOut now:

Sync:  Clock: 231, Tol: 123, PattLen: 7 (7), Pulse: -248, 607, mStart: 254, MCD: 0, mtrunc: 0
Signal: 01234343434125012505052501212121212505012105012121250505050125050125050125012125050125050501234343434125012505052501212121212505012121250505050125050125050125012125050125050501234343434125012505050501212501212121212505012121250505050125050125050125012125.  [254]
Pattern:  P0: 22*[-251] P1: 19*[231] P2: 19*[-613] P3: 0*[840] P4: 0*[-858] P5: 22*[607] P6: 0*[]
p0l1H(vcnt:1:RES:)_while i=4while pdec messageLen=254
p0l1L0s0l1L0H(vcnt:5:RES:)_while i=19while pdec messageLen=254
...
p0H(vcnt:0:RES:)_while i=255while pdec messageLen=254
printOut now:

Sync:  Clock: 231, Tol: 123, PattLen: 7 (7), Pulse: -248, 607, mStart: 254, MCD: 0, mtrunc: 0
Signal: 01234343434125012505052501212121212505012105012121250505050125050125050125012125050125050501234343434125012505052501212121212505012121250505050125050125050125012125050125050501234343434125012505050501212501212121212505012121250505050125050125050125012125.  [254]
Pattern:  P0: 22*[-251] P1: 19*[231] P2: 19*[-613] P3: 0*[840] P4: 0*[-858] P5: 22*[607] P6: 0*[]
p0l1H(vcnt:1:RES:)_while i=4while pdec messageLen=254
...
```

elektron-bbs, 2021-08-04:
Ich mache mal noch eine Anmerkung:
Die Nachrichten, die hier nicht durch die MC-Routine kommen (wenn es wirklich die im Debug-Log ausgegebene ist), stammen von einem Sensor mit Protokoll 84. Die Nachrichten dieses Sensors werden aber sonst zu 99,9 (geschätzt) als MU ausgegeben und von FHEM korrekt dekodiert.

sidey79, 2021-08-04:
Den Fehler habe ich prinzipiell schon lokalisiert. Es läuft die Variable über.
Ich bin nur noch immer am überlegen und tüfteln, wie ich das geschickt behebe.

Mir sind da noch zwei andere Stellen aufgefallen,. die hängen damit zusammen, diese sind auch nicht ganz fehlerfrei optimal.

sidey79, 2021-08-07:
Den Fehler habe ich prinzipiell ausgebaut, allerdings ändert sich das Ergebnis vom mcdecoder, was ich nicht beabsichtige.

Ist ein bisschen tüftelei, weil die Vielzahl an Bedingungen es in sich haben und doch ziemlich komplex machen.

HomeAutoUser, 2021-08-07:
Es ist schonmal gut, zu wissen wo der Fehler liegt. Ein anderes Ergebnis zu verändern ist natürlich nicht zielführend. Vielleicht ergibt sich beim betrachten noch eine Idee um die Komplexität zu „vereinfachen“.

Was ist, wenn man bei i 255 abbrechen würde?

sidey79, 2021-08-07:
Tia, so in der Art will ich es umbauen (einfach ausgedrückt).

Aktuell befindet sich "abbrechen" in einem Else Zweig diverser Bedingungen. (geschachtelt)
Ich möchte das umdrehen und den Abbruch gezielt durch Bedingungen erreichen.

closed 2021-08-30 by HomeAutoUser