RFD-FHEM/SIGNALDuino #263: SignalDuino auf dem Wemos D1 mini (ESP8266) empfängt nichts

djusHa, 2022-12-12:
Hallo,

bekomme SignalDuino nicht auf dem Wemos D1 mini zum laufen.

Benutze die vorkompillierte  SIGNALDuino_ESP8266cc1101_3.5.0.hex.

Bin nach dieser Verkabelung vorgegangen: https://raw.githubusercontent.com/LSatan/SmartRC-CC1101-Driver-Lib/master/img/Esp8266_CC1101.png

Hatte probeweise zwei unterschiedliche CC1101 Module dran ( 433Mhz ), die Module sind ok, auf einem Arduino Nano
laufen diese Einwandfrei.

In FHEM wird alles erkannt, die Attribute sind da und auch richtig, die "gets und sets" funktionieren.

Nur Empfangen tut der nix.

Was kann es sein?

sidey79, 2022-12-12:
Hast Du ihn so verkabelt?

<html>
<body>
<p style="margin: 0.5em 0px; color: rgb(32, 33, 34); font-family: sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">An den "SIGNALESP" kann auch ein CC1101 via SPI angebunden werden:</p>

CC1101 Bezeichnung | ESP Pin
-- | --
CLK | GPIO14
MOSI | GPIO13
MISO | GPIO12
CSN | GPIO15
GDO0 | GPIO4
GDO2 | GPIO5

<br class="Apple-interchange-newline">
</body>
</html>

djusHa, 2022-12-12:
Jap, genauso

elektron-bbs, 2022-12-13:
Wenn du nach dem verlinkten Bild verdrahtet hast, dann sind GDO0 und GDO2 vertauscht.

djusHa, 2022-12-14:
@elektron-bbs 
Stimmt, Du hast recht. Habe die Anschlüsse jetzt gedreht. Ohne Erfolg

elektron-bbs, 2022-12-14:
Eigenartig, wenn set und get funktioniert, bedeutet das, das SPI funktioniert - dann bleiben ja nur noch die beiden GDO übrig. Ich lade mal noch ein PDF von meinem Eigenbau hoch, das mit der Original-Firmware funktioniert.
[SIGNAL-ESP_WeMos-D1-Mini_CC1101_SCH.PDF](https://github.com/RFD-FHEM/SIGNALDuino/files/10230088/SIGNAL-ESP_WeMos-D1-Mini_CC1101_SCH.PDF)

djusHa, 2022-12-16:
Ich mag es gar nicht sagen, aber...

Schuld waren die Steckkabel, habe die Federkontakte jetzt nachgebogen, der SignalDuino rennt jetzt.

Gibts es eigentlich eine Möglichkeit die LED blinken zu lassen sobal der SignalDuino was empfängt oder sendet?

elektron-bbs, 2022-12-16:
:-)
Wann genau die LED angesteuert wird, kann ich auch nicht beantworten. Das kann @sidey79 sicher besser.
In den FSK-Modes blitzt sie auf jeden Fall bei jedem Empfang. Ich habe mir angewöhnt, zwei zusätzliche LEDs zu spendieren, eine für TX und eine für RX (siehe PDF). Man muss nur darauf achten, das die Ausgänge vom CC1101 mit maximal 4 mA belastet werden dürfen.

djusHa, 2022-12-16:
Ok, low power LED's müsste ich noch haben...

Mal eine andere Frage, ich wollte in verbindung mit dem SignalDuino folgende 433Mhz lernfähige Fernbedienungen benutzen:
[Ebay](https://www.ebay.de/itm/374316392976?_trkparms=amclksrc%3DITM%26aid%3D777008%26algo%3DPERSONAL.TOPIC%26ao%3D1%26asc%3D20220822113838%26meid%3Daadf2d54c8114fc3873e778b34f47904%26pid%3D101524%26rk%3D1%26rkt%3D1%26itm%3D374316392976%26pmt%3D0%26noa%3D1%26pg%3D2380057%26algv%3DRecentlyViewedItemsV2%26brand%3DMarkenlos&_trksid=p2380057.c101524.m146925&_trkparms=pageci%3Ac0cd703c-7d78-11ed-96dc-6e9fadacaf9e%7Cparentrq%3A1c6eee721850a45b15d37182ffff5a92%7Ciid%3A1)

Diese Arbeiten mit 433.92 Mhz und sind kompatibel zu den hörmann RSC2 und RSE2. Es müsste doch möglich sein die FB's mit dem Signalduino anzulernen, wenn man mit richtigen Protokoll sendet. Die einzelnen Tasten sind zwar mit einem Code vorbelegt, dieser ist aber immer gleich, so dass man die FB's nicht vernünftig nutzen kann.

Hier der Log, es sind insgesamt 4 Tastendrücke gewesen auf die selbe Taste:
```
2022-12-16 20:44:01 SIGNALduino wlanduino DMSG u19#FEBEE2
2022-12-16 20:44:01 SIGNALduino wlanduino UNKNOWNCODE u19#FEBEE2
2022-12-16 20:44:01 SD_UT unknown_please_select_model ???
2022-12-16 20:44:01 SD_UT unknown_please_select_model unknownMSG: 111111110101111101110001  (protocol: 118)
2022-12-16 20:44:01 SIGNALduino serialduino DMSG u19#FEBEE2
2022-12-16 20:44:03 SIGNALduino wlanduino DMSG u19#FEBEE2
2022-12-16 20:44:03 SIGNALduino wlanduino UNKNOWNCODE u19#FEBEE2
2022-12-16 20:44:03 SD_UT unknown_please_select_model ???
2022-12-16 20:44:03 SD_UT unknown_please_select_model unknownMSG: 111111110101111101110001  (protocol: 118)
2022-12-16 20:44:03 SIGNALduino serialduino DMSG u19#FEBEE2
2022-12-16 20:44:05 SIGNALduino wlanduino DMSG u19#FEBEE2
2022-12-16 20:44:05 SIGNALduino wlanduino UNKNOWNCODE u19#FEBEE2
2022-12-16 20:44:05 SD_UT unknown_please_select_model ???
2022-12-16 20:44:05 SD_UT unknown_please_select_model unknownMSG: 111111110101111101110001  (protocol: 118)
2022-12-16 20:44:05 SIGNALduino serialduino DMSG u19#FEBEE2
2022-12-16 20:44:07 SIGNALduino wlanduino DMSG u19#FEBEE2
2022-12-16 20:44:07 SIGNALduino wlanduino UNKNOWNCODE u19#FEBEE2
2022-12-16 20:44:07 SD_UT unknown_please_select_model ???
2022-12-16 20:44:07 SD_UT unknown_please_select_model unknownMSG: 111111110101111101110001  (protocol: 118)
2022-12-16 20:44:07 SIGNALduino serialduino DMSG u19#FEBEE2
```

Kann man da was Sinnvolles raus machen?

djusHa, 2022-12-16:
Hier nochmal mit Verbose 5 und ein einzelner Tastendruck:
```
2022.12.16 20:53:30 5 : wlanduino: Read, RAW rmsg: Mu;�ъ;���;�փ;��;�Ǆ;��;���;���;DBBRRbRbWb;C2;R76;
2022.12.16 20:53:30 4 : wlanduino: Read, msg READredu: MU;P0=-2641;P1=-429;P2=982;P3=361;P4=-1095;P5=-8292;P6=-11779;P7=246;D=1212121212121213421342125252625262570262;CP=2;R=118;
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 13.1 -> FLAMINGO FA22RF / FA21RF / LM-101LD not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 16 -> Dooya not found, aborting
2022.12.16 20:53:30 4 : wlanduino: Parse_MU, Fingerprint for MU protocol id 19 -> minify matches, trying to demodulate
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, 0. try, regex ((?:)((?:21|34){19,})) did not match
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 20.1 -> RCnoName20 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 21 -> Einhell Garagedoor not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 22 -> HAMULiGHT not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 24 -> Visivo remote not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 26 -> xavax not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 27 -> EFTH-800 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 28 -> IC Ledspot not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 31 -> LTECH not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 32 -> wireless doorbell not found, aborting
2022.12.16 20:53:30 4 : wlanduino: Parse_MU, Fingerprint for MU protocol id 34 -> QUIGG | LIBRA | Mandolyn | Pollin ISOTRONIC matches, trying to demodulate
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, 0. try, regex ((?:3)((?:12|42){19,}(?:4|1)?)) did not match
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 36 -> remote not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 37 -> Bresser 7009994 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 38 -> NC-3911 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, for MU protocol id 39, applying filterfunc SIGNALduino_compPattern
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 39 -> X10 Protocol not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 40 -> Romotec  not found, aborting
2022.12.16 20:53:30 4 : wlanduino: Parse_MU, Fingerprint for MU protocol id 42 -> wireless doorbell matches, trying to demodulate
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, 0. try, regex ((?:212121)((?:21|21){28,})) did not match
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 44 -> BresserTemeo not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 44.1 -> BresserTemeo not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 45 -> Revolt not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 46 -> SKXxxx, GF0x0x not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 48 -> TFA Dostmann not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 49.1 -> GT-9000 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 54 -> TFA 30.3233.01 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 56 -> AC114-xxB not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 59 -> AK-HD-4 not found, aborting
2022.12.16 20:53:30 4 : wlanduino: Parse_MU, Fingerprint for MU protocol id 60 -> WS2000 matches, trying to demodulate
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, 0. try, regex ((?:)((?:34|21){38,}(?:2|3)?)) did not match
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 62 -> Clarus_Switch not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 66 -> WS7035 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 69 -> Hoermann not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 72 -> Siro shutter not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 76 -> LED XM21 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 79 -> wireless doorbell not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 81 -> SA-434-1 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 83 -> RH787T not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 84 -> IAN 283582 / TV-4848 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 85 -> TFA 30.3222.02 not found, aborting
[... cut ...]
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 105 -> BF-301 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 110 -> ADE_WS_1907 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 111 -> TS-FT002 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 114 -> TR401 not found, aborting
2022.12.16 20:53:30 5 : wlanduino: Parse_MU, start pattern for MU protocol id 122 -> TM40 not found, aborting
```

elektron-bbs, 2022-12-17:
Das müssten wir uns genauer ansehen. Bitte eröffne dazu ein neues Issue unter https://github.com/RFD-FHEM/RFFHEM/issues
Wir benötigen dann auf jeden Fall RAW-msg von allen Tasten.

djusHa, 2022-12-17:
Erledigt:
https://github.com/RFD-FHEM/RFFHEM/issues/1134#issue-1501318856

Bin sehr gespannt, die FB ist ziemlich cool um mal was aus dem Auto zu schalten

sidey79, 2023-01-25:
@djusHa 

Ist das Empfangsthema (Wemos) erledigt, dann würde ich den Vorfall gerne schließen

djusHa, 2023-01-26:
Ja, kann zu.
Vielen Dank für deine Zeit!

closed 2023-01-26 by sidey79