RFD-FHEM/SIGNALDuino #264: rc-switch "protocol 1" PT2260, EV1527 support

mightymos, 2023-01-19:
I have written an alternative firmware for a particular door sensor operating at 433 MHz:
https://github.com/mightymos/ReedTripRadio

I chose to transmit using "protocol 1" which is supported by rc-switch and portisch projects.
This allows Sonoff RF Bridge to be used for receiving (along with hardware or firmware modification).

I have one user that could not get my firmware working with SIGNALDuino:
https://github.com/mightymos/ReedTripRadio/issues/4

The timings can be easily read here for "protocol 1":
https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/inc/RF_Protocols.h

I must have poor eyesight, I could not find where SIGNALDuino stores protocol timings?
Is there any interest in supporting an additional protocol?

I will help as best I can if requested.

- Cheers from USA

sidey79, 2023-01-20:
The Signalduino isn`t aware of any protocol. It`s more a pattern which ist recognised.

We process a detected Signal outside of the uC. Currently there is only a per based FHEM Backend which translated the signals into protocols an then into sensors.

mightymos, 2023-01-20:
Okay thank you for the explanation.

closed 2023-01-20 by mightymos