RFD-FHEM/SIGNALDuino #272: ecowitt WH51 Auswertung mehrerer Bodenfeuchtesensoren

Antannah, 2023-04-02:
Hallo, mehrere Sensoren des Typs werden in FHEM nicht verarbeitet.
Ich habe zwei WH51 Bodenfeuchtesensoren gekauft und den Signalduino aus einem WemosD1mini (ESP8266) und einem cc1101 mit 868MHz aufgebaut. Dort habe ich das binary 3.5.4 geflashed und im FHEM die aktuellen updates (1.4.2023) aufgespielt. Signalduino geht prinzipiell auch, die Botschaften werden empfangen und dekodiert. Es wird allerdings nur ein Sensor (SD_WS) ausgewertet.
Sensor S1 hat die ID 0D32C1 (Aufkleber auf dem Sensor) heißt im FHEM se_bodenfeuchteC1
Sensor S2 hat die ID 0D2F91 heißt im FHEM se_bodenfeuchte91
Der SignalDuino in FHEM heißt sduinoESP, dort habe ich "rfmode  Fine_Offset_WH51_868" und "longids 1" eingestellt.

Die Readings von S2 tragen das heutige Datum und scheinen damit ab und zu aktualisiert zu werden. Die Readings von S1 tragen als letztes Datum den Moment, als ich S21 hinzugefügt habe und werden folglich nicht aktualisiert.

Hier das LOG:
`
2023.04.02 10:59:56 4: sduinoESP: KeepAlive, ok, retry = 0
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Error! id 117 msg=510D32C1107F00F837FFFFFFEEFA, message is to short
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, 2-FSK Protocol id 116.1 WH57 msg 510D32C1107F00F837FFFFFFEEFA not match (?^:^57)
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, 2-FSK Protocol id 116 WH57 msg 510D32C1107F00F837FFFFFFEEFA not match (?^:^57)
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Error! id 115 msg=510D32C1107F00F837FFFFFFEEFA, message is to short
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Error! id 112 msg=510D32C1107F00F837FFFFFFEEFA, message is to long
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Error! id 109 msg=510D32C1107F00F837FFFFFFEEFA, message is to long
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Error! id 108 msg=510D32C1107F00F837FFFFFFEEFA, message is to short
2023.04.02 10:59:46 4: sduinoESP: Dispatch, W107#510D32C1107F00F837FFFFFFEEFA, Dropped due to short time or equal msg
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 107.1 -> WH51 868.35 MHz with match (?^:^51)
2023.04.02 10:59:46 3: sduinoESP: se_bodenfeuchte91 ERROR - Hum diff too large (old 54, new 0, diff 54.0)
2023.04.02 10:59:46 4: sduinoESP: using longid for 1 device SD_WS_107_H_0D32C1
2023.04.02 10:59:46 4: sduinoESP: SD_WS_Parse decoded protocol-id 107 (WH51, DP100, MISOL/1), sensor-id 0D32C1
2023.04.02 10:59:46 4: sduinoESP: SD_WS_Parse protocol 107, rawData 510D32C1107F00F837FFFFFFEEFA
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 107 -> WH51 433.92 MHz with match (?^:^51)
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, 2-FSK Protocol id 103 Lacrosse mode 2 msg 510D32C1107F00F837FFFFFFEEFA not match (?^:^9)
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, GFSK Protocol id 102 KoppFreeControl msg 510D32C1107F00F837FFFFFFEEFA not match (?^:^0)
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Error! method ConvPCA301, checksumCalc:8D1E != checksum:FFFF
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 101 -> PCA 301
2023.04.02 10:59:46 4: sduinoESP: Parse_MN, 2-FSK Protocol id 100 Lacrosse mode 1 msg 510D32C1107F00F837FFFFFFEEFA not match (?^:^9)
2023.04.02 10:59:46 4: sduinoESP: Read, msg: MN;D=510D32C1107F00F837FFFFFFEEFA;R=88;
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Error! id 117 msg=510D2F91107F00F83DFFFFFFC09F, message is to short
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, 2-FSK Protocol id 116.1 WH57 msg 510D2F91107F00F83DFFFFFFC09F not match (?^:^57)
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, 2-FSK Protocol id 116 WH57 msg 510D2F91107F00F83DFFFFFFC09F not match (?^:^57)
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Error! id 115 msg=510D2F91107F00F83DFFFFFFC09F, message is to short
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Error! id 112 msg=510D2F91107F00F83DFFFFFFC09F, message is to long
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Error! id 109 msg=510D2F91107F00F83DFFFFFFC09F, message is to long
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Error! id 108 msg=510D2F91107F00F83DFFFFFFC09F, message is to short
2023.04.02 10:59:07 4: sduinoESP: Dispatch, W107#510D2F91107F00F83DFFFFFFC09F, Dropped due to short time or equal msg
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 107.1 -> WH51 868.35 MHz with match (?^:^51)
2023.04.02 10:59:07 3: sduinoESP: se_bodenfeuchte91 ERROR - Hum diff too large (old 54, new 0, diff 54.0)
2023.04.02 10:59:07 4: sduinoESP: using longid for 1 device SD_WS_107_H_0D2F91
2023.04.02 10:59:07 4: sduinoESP: SD_WS_Parse decoded protocol-id 107 (WH51, DP100, MISOL/1), sensor-id 0D2F91
2023.04.02 10:59:07 4: sduinoESP: SD_WS_Parse protocol 107, rawData 510D2F91107F00F83DFFFFFFC09F
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 107 -> WH51 433.92 MHz with match (?^:^51)
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, 2-FSK Protocol id 103 Lacrosse mode 2 msg 510D2F91107F00F83DFFFFFFC09F not match (?^:^9)
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, GFSK Protocol id 102 KoppFreeControl msg 510D2F91107F00F83DFFFFFFC09F not match (?^:^0)
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Error! method ConvPCA301, checksumCalc:B455 != checksum:FFFF
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 101 -> PCA 301
2023.04.02 10:59:07 4: sduinoESP: Parse_MN, 2-FSK Protocol id 100 Lacrosse mode 1 msg 510D2F91107F00F83DFFFFFFC09F not match (?^:^9)
2023.04.02 10:59:07 4: sduinoESP: Read, msg: MN;D=510D2F91107F00F83DFFFFFFC09F;R=80;
2023.04.02 10:58:56 4: sduinoESP: KeepAlive, ok, retry = 0
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Error! id 117 msg=510D32C1107F00F838FFFFFF606D, message is to short
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, 2-FSK Protocol id 116.1 WH57 msg 510D32C1107F00F838FFFFFF606D not match (?^:^57)
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, 2-FSK Protocol id 116 WH57 msg 510D32C1107F00F838FFFFFF606D not match (?^:^57)
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Error! id 115 msg=510D32C1107F00F838FFFFFF606D, message is to short
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Error! id 112 msg=510D32C1107F00F838FFFFFF606D, message is to long
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Error! id 109 msg=510D32C1107F00F838FFFFFF606D, message is to long
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Error! id 108 msg=510D32C1107F00F838FFFFFF606D, message is to short
2023.04.02 10:58:36 4: sduinoESP: Dispatch, W107#510D32C1107F00F838FFFFFF606D, Dropped due to short time or equal msg
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 107.1 -> WH51 868.35 MHz with match (?^:^51)
2023.04.02 10:58:36 3: sduinoESP: se_bodenfeuchte91 ERROR - Hum diff too large (old 54, new 0, diff 54.0)
2023.04.02 10:58:36 4: sduinoESP: using longid for 1 device SD_WS_107_H_0D32C1
2023.04.02 10:58:36 4: sduinoESP: SD_WS_Parse decoded protocol-id 107 (WH51, DP100, MISOL/1), sensor-id 0D32C1
2023.04.02 10:58:36 4: sduinoESP: SD_WS_Parse protocol 107, rawData 510D32C1107F00F838FFFFFF606D
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 107 -> WH51 433.92 MHz with match (?^:^51)
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, 2-FSK Protocol id 103 Lacrosse mode 2 msg 510D32C1107F00F838FFFFFF606D not match (?^:^9)
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, GFSK Protocol id 102 KoppFreeControl msg 510D32C1107F00F838FFFFFF606D not match (?^:^0)
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Error! method ConvPCA301, checksumCalc:AF1E != checksum:FFFF
2023.04.02 10:58:36 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 101 -> PCA 301
[... 68 lines cut ...]
2023.04.02 10:56:16 4: sduinoESP: Parse_MN, Error! id 115 msg=510D32C1107F00F838FFFFFF606D, message is to short
2023.04.02 10:56:16 4: sduinoESP: Parse_MN, Error! id 112 msg=510D32C1107F00F838FFFFFF606D, message is to long
2023.04.02 10:56:16 4: sduinoESP: Parse_MN, Error! id 109 msg=510D32C1107F00F838FFFFFF606D, message is to long
2023.04.02 10:56:16 4: sduinoESP: Parse_MN, Error! id 108 msg=510D32C1107F00F838FFFFFF606D, message is to short
2023.04.02 10:56:16 4: sduinoESP: Dispatch, W107#510D32C1107F00F838FFFFFF606D, Dropped due to short time or equal msg
2023.04.02 10:56:16 4: sduinoESP: Parse_MN, Found 2-FSK Protocol id 107.1 -> WH51 868.35 MHz with match (?^:^51)
2023.04.02 10:56:16 3: sduinoESP: se_bodenfeuchte91 ERROR - Hum diff too large (old 54, new 0, diff 54.0)
2023.04.02 10:56:16 4: sduinoESP: using longid for 1 device SD_WS_107_H_0D32C1
2023.04.02 10:56:16 4: sduinoESP: SD_WS_Parse decoded protocol-id 107 (WH51, DP100, MISOL/1), sensor-id 0D32C1
[... cut ...]
2023.04.02 10:56:16 4: sduinoESP: Read, msg: MN;D=510D32C1107F00F838FFFFFF606D;R=88;
`

Ich erkenne:
S1 wird auf S2 gemapped, kommt mit den dortigen Werten durcheinander, und wird dann gedropped:

'2023.04.02 10:56:16 3: sduinoESP: se_bodenfeuchte91 ERROR - Hum diff too large (old 54, new 0, diff 54.0)
2023.04.02 10:56:16 4: sduinoESP: using longid for 1 device SD_WS_107_H_0D32C1
2023.04.02 10:56:16 4: sduinoESP: Dispatch, W107#510D32C1107F00F838FFFFFF606D, Dropped due to short time or equal msg'

elektron-bbs, 2023-04-02:
Tja, kann ich nicht nachvollziehen, da bei allen Nachrichten der Feuchtewert gleich 0 ist.
Wieso jeweils die Meldung
`se_bodenfeuchte91 ERROR - Hum diff too large (old 54, new 0, diff 54.0)`
erscheint, ist mir schleierhaft.

Poste bitte mal ein list beider Sensoren.

Antannah, 2023-04-02:
define se_bodenfeuchteC1 SD_WS SD_WS_107_H_0D32C1
setuuid se_bodenfeuchteC1 642854c3-f33f-c50d-9e8a-34ac0e9eba80b1ee
attr se_bodenfeuchteC1 group Feuchte
attr se_bodenfeuchteC1 icon weather_rain_gauge
attr se_bodenfeuchteC1 room Bewaesserung,Blumen
define se_bodenfeuchte91 SD_WS SD_WS_107_H_0D2F91
setuuid se_bodenfeuchte91 64285ce1-f33f-c50d-1790-7b280a76decde256
attr se_bodenfeuchte91 group Feuchte
attr se_bodenfeuchte91 icon weather_rain_gauge
attr se_bodenfeuchte91 room Bewaesserung,Blumen