RFD-FHEM/SIGNALDuino #281: Advantek Doorbell(Batterie / Funk) wird nicht dekodiert

rcmcronny, 2023-05-10:
Hallo,

ich habe mir eine neue Funkklingel zulegen müssen und Advantek steht zwar bei untersützt mit da, allerdings nur die Steckdosen Variante wohl. Ich habe die Variante mit Batterien für den Empfänger.  (Unten der Amazon Link)

Der SignalESP erkennt was, siehe unten, kann es aber wohl nicht dekodieren, vielleicht ist das ja machbar (oder halt nicht).

Version: SignalESP - Version V 3.5.0 SIGNALESP cc1101 (chip CC1101) - compiled at Jul 8 2022 18:09:41
Daten des SignalESPs / FHEM
```
cc1101_config | Freq: 433.920 MHz, Bandwidth: 325 kHz, rAmpl: 42 dB, sens: 8 dB, DataRate: 5.60 kBaud
cc1101_config_ext | Modulation: ASK/OOK 
cc1101_patable | C3E = 00 84 00 00 00 00 00 00 => 5_dBm
config | MS=1;MU=1;MC=1;Mred=1
versionProtocols  1.48
versionmodul 3.5.4
```
```
2023-05-10 20:03:40-MU;P0=106;P1=-188;P2=-100;P3=216;P5=292;D=0102013202015101010101320132010101013201013201013201013;CP=0;R=26;
#2023-05-10 20:03:40-MU;P0=222;P1=-187;P2=101;P3=-97;P5=298;D=01232301232123210323215121212121032103212121210321210;CP=2;R=26;
#2023-05-10 20:03:40-MU;P0=205;P1=-192;P2=101;P3=-101;P4=299;D=01232123210323214121212121032103212121210321210321210;CP=2;R=28;
#2023-05-10 20:03:40-MU;P0=102;P1=-97;P3=-186;P4=318;P5=212;D=01034303030303510351030303035103035103035103035;CP=0;R=29;
#2023-05-10 20:03:41-MU;P0=221;P1=-191;P2=103;P3=-97;P5=296;D=01232301232123210323215121212121032103212121210321210;CP=2;R=28;
```

Amazon Link https://www.amazon.de/dp/B07K1613FZ

Danke,
Ronny

elektron-bbs, 2023-05-11:
Die Variante von Avantek, die wir bisher eingebaut haben, sendet mit FSK-Modulation und ist nicht kompatibel zu deiner.

Die Nachrichten, die gepostet hast, sind alle zu kurz und wahrscheinlich unvollständig:

```
2023-05-10 20:03:40-MU;P0=106;P1=-188;P2=-100;P3=216;P5=292;D=
010201320201
51
01010101320132010101013201013201013201013;CP=0;R=26;

2023-05-10 20:03:40-MU;P0=222;P1=-187;P2=101;P3=-97;P5=298;D=
0123230123212321032321
51
21212121032103212121210321210;CP=2;R=26;

2023-05-10 20:03:40-MU;P0=205;P1=-192;P2=101;P3=-101;P4=299;D=
0123212321032321
41
21212121032103212121210321210321210;CP=2;R=28;

2023-05-10 20:03:40-MU;P0=102;P1=-97;P3=-186;P4=318;P5=212;D=
0103
43
03030303510351030303035103035103035103035;CP=0;R=29;

2023-05-10 20:03:41-MU;P0=221;P1=-191;P2=103;P3=-97;P5=296;D=
0123230123212321032321
51
21212121032103212121210321210;CP=2;R=28;
```

Es könnte sein, das die Pulse (ca. 100 µS)  zu kurz sind für den SIGNALduino.

rcmcronny, 2023-05-11:
Hi,
danke für die Info. 
Optisch paßt es genau zu der Analyse:  http://techbotch.org/blog/wireless-doorbell/index.html

Lässt sich da was machen, oder ist es generell nicht machbar (wäre ok für mich)

Danke Ronny