RFD-FHEM/SIGNALDuino #286: Setting up the project and compiling

felixxxxxs, 2023-05-20:
It would be very nice if there would be more detailed instructions on how to setup the project in one of the IDEs (preferrably Arduino IDE, or VS Code or Visual Studio) and how to compile manually for a specific board. Unfortunately, the short paragrah in the Readme doesn't help me much.

Coming from Windows and a completely different stack, I'm already struggling with cloning the repo because of the symbolic links. :(

sidey79, 2023-05-20:
I'am currently improving the VSCode setup:
https://github.com/RFD-FHEM/SIGNALDuino/pull/278

1. you have to clone the repo. This shouldn't be a problem on windows or linux. If it is, then tell me more about the problem please.
The symbolic links are only used for the arduino ide, so that all files are in the same folder.
2. If you use vs code, then you can open the project and compile via platformio.
3. Visual Studio need the visual micro addon, thich i use'd years ago.

felixxxxxs, 2023-05-21:
Ich antworte jetzt einfach mal auf Deutsch, danke für die schnelle Antwort!

Das Repo klonen ist kein Problem, nur die Symlinks funktionieren unter Windows nicht, daher fällt für mich die Arduino IDE weg.

Ich habe das Projekt in VS Code geöffnet. Ich habe die C/C++ Extensions installiert sowie die PlatformIO IDE.
Es kommen sofort eine Reihe an Fehlern, darunter, dass der Arduino.h include fehlt:
[image]

Meine .vscode/c_cpp_properties.json sieht folgendermaßen aus (ist die Standardkonfiguration):
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}/**"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "compilerPath": "C:\\msys64\\mingw64\\bin\\gcc.exe",
            "cStandard": "c17",
            "cppStandard": "gnu++17"
        }
    ],
    "version": 4
}
```
Wie sieht deine Konfiguration aus? 😃 

Außerdem kommt folgende Fehlermeldung, wenn ich versuche, über PlatformIO bspw. für esp8266_cc1101 zu builden:
[image]
Was mich wundert, da er sich ja offensichtlich über die Namen in der platformio.ini beschwert, was er dann ja auch bei anderen Leuten machen sollte.

sidey79, 2023-05-21:
> Wie sieht deine Konfiguration aus? 😃
>
> Außerdem kommt folgende Fehlermeldung, wenn ich versuche, über PlatformIO bspw. für esp8266_cc1101 zu builden: [image] Was mich wundert, da er sich ja [...]

Ja, das Problem existiert seit 8.5.23 , weil mit der Version 6.1.7 eine Prüfung auf die Namen eingefügt wurde. https://docs.platformio.org/en/latest/core/history.html#id2

Ich nutze deshalb noch 6.1.6 da ich noch nicht zum Umbenennen kam.

felixxxxxs, 2023-05-21:
Ich habe folgende Schritte durchgeführt:
- In der platformio.ini @ durch - ersetzt
- ArduinoJson Library über PlatformIO installiert
- WiFiManager manuell installiert, da die Version in PlatformIO veraltet ist

Jetzt funktionieren meine Includes, und ich konnte die Firmware erfolgreich kompilieren. Vielen Dank!

Gibt es einen Grund, warum die Libraries ArduinoJson und WiFiManager nicht mitgeliefert werden?

Wenn gewünscht, kann ich schnell einen PR mit den angepassten Environment Namen erstellen.

closed 2023-06-09 by sidey79