RFD-FHEM/SIGNALDuino #290: minicul PIN_MARK433

elektron-bbs, 2023-06-24:
Kann es sein, das für den minicul noch das Flag PIN_MARK433 gesetzt werden muss, siehe https://forum.fhem.de/index.php?msg=1279688?
Evtl. trifft das auch auf den radino zu. Ich kann es nicht nachvollziehen, da ich nur einen radino für 433 MHz besitze.

sidey79, 2023-06-24:
Ich hab nachgesehen, für den mincul wird PIN_MARK433 auf A0 verwendet.

Es wird ein pullup gemacht, bedeutet also der pin ist high bei der 868 Version.
Laut Schaltplan ist bei der 433 Version der pin mit GND verbunden.

Vielleicht hält sich da einfach jemand nicht an die HW Spec?

elektron-bbs, 2023-06-25:
> Vielleicht hält sich da einfach jemand nicht an die HW Spec?

Sieht so aus, als hätte der Hersteller versehentlich die Brücke an Pin A0 gegen GND gesetzt.

Mir ist aber dabei noch etwas anderes aufgefallen. Bei der Version Version 3.5.1 RC1 wird offensichtlich für den minicullCC1101 kein CC1101 mit compiliert. Die Dateigröße lässt darauf schließen:

```
Version 3.5.1 RC1
SIGNALduino_miniculCC1101_3.5.1-RC1.hex   58.8 KB   last week
SIGNALduino_nano328s_3.5.1-RC1.hex        58.9 KB   last week
SIGNALduino_nanoCC1101_3.5.1-RC1.hex      67.8 KB   last week

Release 3.5.0
SIGNALDuino_miniculcc1101_3.5.0.hex       67.4 KB   Jul 8, 2022
SIGNALDuino_nano328_3.5.0.hex             59   KB   Jul 8, 2022
SIGNALDuino_nanocc1101_3.5.0.hex          67.8 KB   Jul 8, 2022
```

closed 2025-01-19 by elektron-bbs