RFD-FHEM/SIGNALDuino #293: Siral EL4F motor with 433 MHz remote control

8cH9azbsFifZ, 2023-07-28:
Hi there. I am trying to use a minicul with signalduino for controlling my siral motor. I got the protocol recorded using an rtl-sdr and a replay using raspi is working (https://github.com/8cH9azbsFifZ/schanz-rolladen-raspi). 

rtl_433 -A -r 250k:433.95M:cu8:button_open.iq
Yields: Use a flex decoder with -X 'n=name,m=OOK_PWM,s=332,l=2064,r=15748,g=2004,t=693,y=0'

[04] {18} 15 1d 40 : 00010101 00011101 01

This is what I receive with the minicul. What can I do?

`Mu;???;???;???;??;???;???;?ց;???;D4TTVtVtVtTTVvvtVvvTTVtVtVtTTVvvtVvvTTVtVtVtTTVvvtVvvTTVtVtVtTTVvvtVvvTTVtVtVtTTVvvtVvvTTVtVtVtTTVvvtVvvTTVtVtVtTTVvvtVvv;C6;R19;O;
                                                                                                                                                                    Mu;???;???;???;?ρ;???;D!!#A#A#A!!#CCA#CC!!#A#A#A!!#CCA#CC!!#A#A#A!!#CCA#CC!!#A#A#A!!#CCA#CC!!#A#A#A!!#CCA#CC!!#A#A#A!!#CCA#CC!!#A#A#A!!#CCA#CC;C3;R18;O;
                                                                                                                                                     Mu;???;???;?ҁ;???;???;D11122122A11122122A11122122A11122122A11122122A11122122A11122122A;C2;R18;O;
                                                                              Mu;???;???;?ԁ;???;???;D11122122A11122122A11122122A11122122A11122122A11122122A11122122A;C2;R15;O;
       Mu;???;?ā;???;???;???;???;D###!!#!!C###!!#!!C###!!#!!C###!!#!!C###!!#!!C###!!#!!C##;C1;R18;`

sidey79, 2023-07-28:
Connect minicul to FHEM, then you geht a more redable Repräsentation of the data

8cH9azbsFifZ, 2023-08-11:
How I can receive this message: 

2023.08.11 14:20:30.870 1: sigduino: SD_UT_Parse UNDEFINED sensor unknown detected, protocol 46, data EAE28, code 1FFF1F0

How could I replay this signal?

elektron-bbs, 2023-08-11:
Ich habe das Vorgehen hier beschrieben: https://forum.fhem.de/index.php?msg=925513
Das Update auf die Entwicklerversion ist jetzt nicht mehr notwendig, da das Protokoll mittlerweile im SVN ist.

8cH9azbsFifZ, 2023-08-11:
Integrated it as a first hack in my scripts and the rollershutter works with these commands:

  
  def _press_button_open(self):
        open_command = "set sigduino sendMsg P46#111010101110001010#R10"
        self._fhem.send_cmd(open_command) 
        
    def _press_button_close(self):
        close_command = "set sigduino sendMsg P46#111010101110001000#R10"
        self._fhem.send_cmd(close_command)

closed 2025-08-16 by elektron-bbs