RFD-FHEM/SIGNALDuino #306: Receive correct signals with current version 3.5.0

budda85, 2023-11-13:
Hello,
I wanted to update my Signalduino and have flashed the current hex file. After that, I can not longer receive or send correct signals. Has anything changed here?
I have flashed with the following command:
`avrdude -c arduino -P /dev/ttyUSB0 -b 57600 -p atmega328p -D -U flash:w:SIGNALDuino_nano328_3.5.0.hex:i`
The only version that shows me something proper is 3.3.0
I have been using the Siganlduino with FHEM for a long time and wanted to connect it to a Raspberrymatic with CUxD. With version 3.3.0 I can at least turn on the switches. Unfortunately, switching them off doesn't work here, but that could also be due to the old Intertechno switch.
Have I forgotten something that I need to set?
The SIGNALduino is built with an FS1000A and RXB6 module.

Version 3.3.0:
[image]

larger than Version 3.3.0 
[image]

sidey79, 2023-11-13:
The second output is not an general error, it uses a compressed message format, which is not human readable

budda85, 2023-11-13:
Ah, okay. Thx for the quick response. 
Do you know if it is possible to use SIGNALduino at Raspberrymatic or is it only for FHEM?

budda85, 2023-11-13:
Simply connected via USB and then sent a command e.g. `is000000000FFFFF` for an Intertechno socket via the signal duino

sidey79, 2023-11-13:
You can disable message reduction via Serial Command "CDR" or enable it again via "CER".

Sending intertechno is possible also, but you have to provide the raw message to the uC, because the conversion from your command to the raw command is done in the perl module withhin FHEM.

The raw command looks somehow like this one, but this is not really a intertechno message:
` SR;R=3;P0=1230;P1=-3120;P2=-400;P3=-900;D=030301010101010202020202020101010102020202010101010202010120202;`

To get your commands, i recommend to install fhem and capture the raw commands.