RFD-FHEM/SIGNALDuino #321: ccmode 0 setzen?

sidey79, 2024-01-03:
@HomeAutoUser @elektron-bbs 

Ich bin am rätseln, wie der ccmode auf 0 Gesetz wird.

Vielleicht habt ihr ja noch eine Erinnerung wieso das hier in 0x18 gesucht wird, aber im Kommentar 0x12 MDMCFG angegeben wurde.

https://github.com/RFD-FHEM/SIGNALDuino/blob/2d7a75c17fbabd3f2cdb20ce80151af0dd82c93d/src/cc1101.cpp#L542C7-L542C7

Ich Versuche aktuell die PCA301 Dosen zum Laufen zu bekommen und es klappt halt nicht.

sidey79, 2024-01-03:
Okay, war dann gestern doch zu spät. 0x12 = 18 :)

closed 2024-01-03 by sidey79

HomeAutoUser, 2024-01-03:
Solltest du herumspielen mit den Registern, so empfehlen wir dir gern https://github.com/HomeAutoUser/cc1101_rf_Gateway 😉

sidey79, 2024-01-03:
Interessant, aber wenn ich das richtig verstanden habe, dan muss ich ein paring Datensatz senden, bevor die Steckdose was sendet.

Kann man hiermit senden?

HomeAutoUser, 2024-01-03:
Senden sollten wir eingebaut haben :) gern hilft da auch @elektron-bbs weiter. 

Hier geht das https://github.com/HomeAutoUser/cc1101_rf_Gateway/blob/master/screenshots/picture5.png

Wir haben das Projekt ja signalduino compatible gestaltet. Uns hat er schon viel geholfen und ist auch in Verbindung mit dem Signalduino Modul aktiv.

Es gibt auch einige Arduino Sketchs für die PCA301. Leider haben wir nicht so eine zum testen.