RFD-FHEM/SIGNALDuino #364: releasenotes version 4.0.0

sidey79, 2025-01-18:
@elektron-bbs 

Ich habe das Release 4.0.0 mal angelegt:
https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/untagged-4e0e287bc9669f8d9cc2

Magst Du noch ein wenig an den Releasenotes wesentliches ergänzen oder hervorherben?

elektron-bbs, 2025-01-18:
Ich würde eigentlich nur folgendes korrigieren wollen:

`Extension for receiving WMBus messages`
in
`Extension for receiving WMBus messages (ESP8266 and ESP32 only)`

und
`added FREQEST additional output`
in
`Automatic Frequency Control (AFC) added to FSK modulation`

Ansonsten finde ich nur die Ausgabe der Version in FHEM etwas "eigenartig" (die beiden V v):

`V v4.0.0 SIGNALESP cc1101 (chip CC1101) - compiled at Jan 18 2025 17:00:28
`

sidey79, 2025-01-19:
@elektron-bbs 

Ah okay, das mit der Version liegt daran, dass ich das Tag als v4.0.0 und nicht 4.0.0 angelegt habe.
Ich hab das jetzt noch mal neu gemacht

Das Release ist hier:
https://github.com/RFD-FHEM/SIGNALDuino/releases/tag/untagged-e51d267f732d4870168c

PS: Kannst Du die Beschreibung da auch ändern oder fehlen da Rechte?

elektron-bbs, 2025-01-19:
Die Versionsausgabe sieht jetzt gut aus.

Die Beschreibung der Releases zu ändern, hatte ich bisher nicht versucht. Das scheint aber zu gehen.

Gibst du Release 4.0.0 dann auch offiziell frei?

sidey79, 2025-01-19:
ja, habe es veröffentlicht