RFD-FHEM/SIGNALDuino #373: ESP32 mit 4 CC1101

elektron-bbs, 2025-07-06:
@sidey79 
Ich würde jetzt gern die Version mit Unterstützung von bis zu 4 CC110x am ESP32 veröffentlichen.
Es sind auch noch einige zusätzliche Änderungen nebenbei mit eingeflossen:

- Maple Mini Unterstüzung WMBus
- WiFi Konfiguration
- ArduinoJson entfernt
- Fastdelegate entfernt
- cronjob geändert

Da die Änderungen ziemlich umfangreich sind, wirst du das sicher nicht in einem Pull request haben wollen.
Soll ich versuchen, das aufzusplitten, oder reicht ein neuer Branch?

elektron-bbs, 2025-07-06:
Das wird etwas schwierig.
Ich könnte versuchen, erst mal abzutrennen, was nicht direkt mit dem ESP32 zu tun hat.
Vorbereitet habe ich jetzt erst mal das Entfernen von ArduinoJson.
Dann vielleicht weiter:

fastdelegate entfernen
Maple Mini WMBus
WiFi

Bei den cronjobs steige ich nicht dahinter, warum teilweise das Intervall geändert wird. Z.B. in der signalesp.h:

```
  long timerTime = maxPulse - duration + 1000;
  if (timerTime < 1000)
    timerTime=1000;

#ifdef ESP32
  esp_timer_stop(cronTimer_handle);
  esp_timer_start_periodic(cronTimer_handle, timerTime);
#elif defined(ESP8266)
  os_timer_disarm(&cronTimer);
  os_timer_arm(&cronTimer, timerTime / 1000, true);
#endif
```

Wozu soll das gut sein? M.E. ist das nicht notwendig.

sidey79, 2025-07-06:
> Das wird etwas schwierig. Ich könnte versuchen, erst mal abzutrennen, was nicht direkt mit dem ESP32 zu tun hat. Vorbereitet habe ich jetzt erst mal das Entfernen von ArduinoJson. Dann vielleicht weiter:
Ja so könntest Du es machen. ArduinoJson brauchte man glaube ich für den WIFIManager wenn man dort Einstellungen speichern wollte. Das hatte ich mal gedacht, für statische IP Adresse aber es funktionierte leider nicht.
 
> fastdelegate entfernen Maple Mini WMBus WiFi

Ja, bin gespannt wieso Du fastdelegate entfernen möchtest.

Große Code Reviews werde ich ohnehin erst nach meinem Urlaub schaffen, also so in ca. 3 Wochen.

> Bei den cronjobs steige ich nicht dahinter, warum teilweise das Intervall geändert wird. Z.B. in der signalesp.h:
> Wozu soll das gut sein? M.E. ist das nicht notwendig.

Der cronjob legt in den maxPulse Wert im FIFO ab, sobald die letzte Pegeländerung > maxPulse ist. Damit wird vermieden auf das Rauschen zu warten. Soweit ich mich erinnere war das auch zum Unterscheiden von Wiederholungen gut.
Der cronjob soll aber frühestens in einer Sekunde wieder aufgerufen werden, egal wie viel Zeit bis maxPulse ist.

elektron-bbs, 2025-07-14:
> Ja, bin gespannt wieso Du fastdelegate entfernen möchtest.

Das nennt sich zwar **fast**delegate, aber ich denke eine direkte Verarbeitung ist trotzdem schneller, zumal nur zwei Funktionen der Bibliothek genutzt wurden. Außerdem wurden in der Arduino-IDE jede Menge Warnungen ausgeworfen. Beim Nano spart die Aktion ca. 1 kB Flash.

> Der cronjob legt in den maxPulse Wert im FIFO ab, sobald die letzte Pegeländerung > maxPulse ist. Damit wird vermieden auf das Rauschen zu warten. Soweit ich mich erinnere war das auch zum [...]

Ich habe mir das jetzt beim Nano genauer angesehen und glaube den Sinn und Zweck verstanden zu haben.

```
  Timer1.setPeriod(32001);

  if (duration >= maxPulse) { //Auf Maximalwert pruefen.
    int sDuration = maxPulse;
    if (isLow(PIN_RECEIVE)) { // Wenn jetzt low ist, ist auch weiterhin low
      sDuration = -sDuration;
    }
    FiFo.enqueue(sDuration);
    lastTime = micros();
  } else if (duration > 10000) {
    Timer1.setPeriod(maxPulse-duration+16);
  }
```

Falls maxPulse noch nicht erreicht wurde, aber der Abstand > 10 **mS** ist, soll der cronjob häufiger ausgeführt werden.
Ich habe das jetzt heraus genommen und sehe eigentlich keinen Nachteil. Im Extremfall wird maxPulse halt erst nach 64 **mS** in den FIFO geschrieben.

elektron-bbs, 2026-01-07:
Ein Stückchen weiter sind wir ja schon. An mir liegt es nicht :-)
https://github.com/RFD-FHEM/SIGNALDuino/pull/392

elektron-bbs, 2026-01-09:
Danke für das Übernehmen der Änderungen. Dann kann ich jetzt mal sehen, womit ich weiter mache.

HomeAutoUser, 2026-11-02:
@elektron-bbs, gibt es hier schon ein Weiterkommen? Ich habe das Projekt noch im Hinterkopf.
Wäre es eigentlich möglich, das dir bekannte Projekt von mir mit 3 Empfängern via "Drahtbrücken" umzustellen, so das man die Hardware uC´s überbrücken würde? *so ne fixe Idee*

elektron-bbs, 2026-11-02:
Ich kam da nicht so richtig weiter, weil @sidey79 und mein Konzept nicht so richtig zusammen passen. Außerdem war mir noch die USB-Buchse vom ESP32 abgebrochen und ich habe mich dann nicht getraut, weitere Experimente durchzuführen.
Beim rfGateway macht der Anschluss von mehreren CC1101 sicher nicht viel Sinn. Das Teil ist auf das Umschalten der Modes konzipiert.

sidey79, 2026-11-02:
Was hat da nicht gepasst?

elektron-bbs, 2026-11-03:
Irgend etwas bei der Ausgabe der Nachrichten. Ich muss mich erstmal wieder hineinvertiefen, um das nachzuvollziehen.