RFD-FHEM/SIGNALDuino #377: v4.0.0 flash problem (esp8266 mit cc1101)

lattemacchiato, 2025-08-14:
Hallo,

der Fehler liegt vermutlich bei mir, aber ich komme nicht weiter:
- D1mini mit CC1101, in fhem als SignalESP
- flashen mit esptool (sudo esptool --port /dev/ttyUSB0 write_flash -fm dout 0x00000 <file>)
- v3.5.0 flashen --> SignalESP läuft & empfängt
- v4.0.0 flashen --> D1mini bleibt "tot", auch per ping nicht erreichbar :-(
- muss ich die bin der v4.0.0. anders flashen als die hex der 3.5.0?
Danke...

lattemacchiato, 2025-08-14:
Danke für den schnellen Tipp! 
0x1000 habe ich auch schon probiert, ändert nichts :-(
Mit "web-Oberfläche" meinst Du fhem? Geht leider beim SignalESP nicht, fhem verweist auf esptool...
Warum gibt's bei der 3.5.0 ein hex file, bei der 4.0.0 ein bin? Ich finde leider nichts dazu, Sorry wenn ich was übersehen...

sidey79, 2025-08-15:
Die Dateiendung wurde nur umbenannt um ein einheitliches Verhalten zu erreichen.

Kannst Du die genauen Dateinamen einmal benennen die Du da im flash befiehlt verwendest?

Eine andere Idee ist, dass auf dem wemosd1mini ein 8255 und kein 8266 verbaut ist.
Der ist mit dem Flash Speicher anders verbunden und braucht einen weiteren Parameter

lattemacchiato, 2025-08-15:
Hab ich - tuts nicht, kein AP. Beim zurück flashen auf 3.5 sind die WLAN Einstellungen wieder wie vorher konfiguriert. 
Ich bin da kein Experte, aber kann es sein, dass im bin der bootloader fehlt? Das würde für ein flashen ab 0x1 sprechen, der bootloader sitzt vermutlich ab 0x0, oder?

lattemacchiato, 2025-08-15:
Hi,

version zeigt:  4.0.0 SIGNALESP cc1101 (chip CC110 unknown) - compiled at Jan 19 2025 13:26:38
(das "unknown" fehlt beim funktionierenden SIGNALESP...)

"raw e" ändert nichts :-(

Der cc1101 wird auf dem "produktiven" D1 als "(chip CC1101)" erkannt, also ohne "unknown".

Muss also am D1 liegen - ich löte mal schnell einen weiteren...

lattemacchiato, 2025-08-15:
Hm, lag wohl am D1 - mit dem neu gelöteten gehts.
Sorry, und Danke für die Unterstützung!

closed 2025-08-17 by sidey79