RFD-FHEM/SIGNALDuino #380: SD_WS_125 - Receive stopped after some times

HomeAutoUser, 2025-08-24:
Problem wurde schon in anderem Faden angesprochen. Nun war es wieder soweit bei mir und ich habe versucht mit dem Handy mehr Infos zu erhalten (weil nicht daheim) -> Receiver blieb wieder stehen.
- Ping erfolgt noch
- state: opened
- nach einem Reset oder Versionsabfrage geht es wieder

Zustände Register beim Nicht-Empfang:
```
0x32 FREQEST - 0хEB
0x35 MARCSTATE - 0x0D
0x38 PKTSTATUS - 0xD0
0x3A TXBYTES - 0x00
0x3B RXBYTES - 0x00
```

Letzte RAWMSG
```
MN;D=30BAA2683190B56C0F0200;R=254;A=5;
RSSI -75
```

elektron-bbs, 2025-08-25:
Die Register sehen eigentlich gut aus:

```
0x32 FREQEST   - 0хEB = -24 = -38,0859375 kHz - hier ähnlich, wackelt 0xE0, 0xE1 ...
0x35 MARCSTATE - 0x0D = RX
0x38 PKTSTATUS - 0xD0 - hier auch so
0x3A TXBYTES   - 0x00 - Number of bytes in TX FIFO = OK
0x3B RXBYTES   - 0x00 - Number of bytes in RX FIFO = OK
```

Ich hatte mir bei der Analyse der DNT_000005 etwas abweichende Werte notiert:

```
Deviation: ca. 70 kHz
Datenrate:
 10 Bit   584 µS =               17,12 kHz
 20 Bit  1110 µS =               18,01 kHz
 30 Bit  1690 µS =               17,75 kHz
352 Bit 10100 µS = 28,7 µS/Bit = 17,42 kHz
-------------------------------------------
                   Durchschnitt: 17,575 kHz
```

Zur Zeit werden die entsprechenden Register so eingestellt:

```
Datenrate: 17.2577, Bandbreite: 135.416667, Deviation: 34.912109
add.  name       def.   cur.
0x10  MDMCFG4  - 0x57 - 0xA9
0x11  MDMCFG3  - 0xC4 - 0x5C
0x15  DEVIATN  - 0x00 - 0x43
```

Probiere mal bitte folgende Einstellungen:

```
Datenrate: 17.5552, Bandbreite: 162.500000, Deviation: 69.824219
add.  name       def.   cur.
0x10  MDMCFG4  - 0x57 - 0x99
0x11  MDMCFG3  - 0xC4 - 0x62
0x15  DEVIATN  - 0x00 - 0x53
```

sidey79, 2025-08-25:
Hmm, ich müsste mal einen Signaldino aufsetzen und dann debuggen... Aber ich weiss noch nicht wie ich das mit dem debuggen anstelle

HomeAutoUser, 2025-08-25:
@elektron-bbs 
Ich habe soeben die Register mal geändert und ein ccconf gemacht. Der Empfang geht wieder, aber es werden immer falsche Chk-Summen gebildet.

```
2025.08.25 21:21:55 3: sduino_868: SD_WS_125 Parse msg 30F8E36B1CCD09B607C180 - ERROR CRC8 73 shoud be 0
2025.08.25 21:21:55 3: sduino_868: SD_WS_125 Parse msg 30F8C32B1CCD09B6078080 - ERROR CRC8 203 shoud be 0
2025.08.25 21:22:03 3: sduino_868: SD_WS_125 Parse msg 30BAF37717D0E13607C180 - ERROR CRC8 237 shoud be 0
2025.08.25 21:22:03 3: sduino_868: SD_WS_125 Parse msg 30BAA33317D0E136078080 - ERROR CRC8 238 shoud be 0
2025.08.25 21:22:08 3: sduino_868: SD_WS_125 Parse msg 3097DB709B09213607C180 - ERROR CRC8 115 shoud be 0
2025.08.25 21:22:08 3: sduino_868: SD_WS_125 Parse msg 3097D3309B092136078080 - ERROR CRC8 157 shoud be 0
2025.08.25 21:22:11 3: sduino_868: SD_WS_125 Parse msg 3050DB671C8450B607C180 - ERROR CRC8 166 shoud be 0
2025.08.25 21:22:11 3: sduino_868: SD_WS_125 Parse msg 305091271C8450B6078080 - ERROR CRC8 175 shoud be 0
2025.08.25 21:22:12 3: sduino_868: SD_WS_125 Parse msg 3056837022C89F3607C080 - ERROR CRC8 50 shoud be 0
2025.08.25 21:22:12 3: sduino_868: SD_WS_125 Parse msg 3056813022C89F3607C080 - ERROR CRC8 199 shoud be 0
2025.08.25 21:22:37 3: sduino_868: SD_WS_125 Parse msg 30FABB7B1AE79B3607C180 - ERROR CRC8 185 shoud be 0
2025.08.25 21:22:37 3: sduino_868: SD_WS_125 Parse msg 30FAB12B1AE79B36078080 - ERROR CRC8 244 shoud be 0
2025.08.25 21:23:00 3: sduino_868: SD_WS_125 Parse msg 30F8E36B1CCD09B607C180 - ERROR CRC8 73 shoud be 0
2025.08.25 21:23:00 3: sduino_868: SD_WS_125 Parse msg 30F8C32B1CCD09B6078080 - ERROR CRC8 203 shoud be 0
2025.08.25 21:23:06 3: sduino_868: SD_WS_125 Parse msg 30BAA33317D0E13607C080 - ERROR CRC8 238 shoud be 0
2025.08.25 21:23:13 3: sduino_868: SD_WS_125 Parse msg 30509B671C8450B607C180 - ERROR CRC8 70 shoud be 0
2025.08.25 21:23:13 3: sduino_868: SD_WS_125 Parse msg 305091271C8450B6078080 - ERROR CRC8 175 shoud be 0
2025.08.25 21:23:13 3: sduino_868: SD_WS_125 Parse msg 3056837022C89F36078180 - ERROR CRC8 50 shoud be 0
2025.08.25 21:23:13 3: sduino_868: SD_WS_125 Parse msg 3056813022C89F36078080 - ERROR CRC8 199 shoud be 0
2025.08.25 21:23:14 3: sduino_868: SD_WS_125 Parse msg 3097D3709B092136078180 - ERROR CRC8 111 shoud be 0
2025.08.25 21:23:14 3: sduino_868: SD_WS_125 Parse msg 3097D3309B092136078080 - ERROR CRC8 157 shoud be 0
```

Aktuelle Config
```
cc1101_config
Freq: 868.370 MHz, Bandwidth: 162 kHz, rAmpl: 33 dB, sens: 8 dB, DataRate: 17.56 kBaud
2025-08-25 21:21:53
cc1101_config_ext
Modulation: 2-FSK, Syncmod: 16/16 sync word bits detected, Deviation: 69.82 kHz
```

Nachdem ich die Register auf rfmode `Fine_Offset_WH31_868` wechselte, klappte der Empfang mit richtigen Chk-Summen auch.

Jetzige Config
```
cc1101_config
Freq: 868.350 MHz, Bandwidth: 135 kHz, rAmpl: 33 dB, sens: 8 dB, DataRate: 17.26 kBaud
2025-08-25 21:29:40
cc1101_config_ext
Modulation: 2-FSK, Syncmod: 16/16 sync word bits detected, Deviation: 34.91 kHz
2025-08-25 21:29:40
```

elektron-bbs, 2026-01-07:
> Hallo [@sidey79](https://github.com/sidey79), [@elektron-bbs](https://github.com/elektron-bbs) -> auf diesem Wege noch ein gesundes neues Jahr 2026

Das wünsche ich euch und allen anderen auch.

> Ich versuche mal ein paar Erkenntnisse hier zu sammeln, da das Problem immer noch besteht. Nach langer Testreihe, habe ich den Verdacht, das vielleicht ein Timer in der uC das Ganze erscheinen lässt. Sobald ich in den Tests mit allen 6 Sensoren immer nach 60 Minuten den uC zurücksetze mit einem RESET läuft es beständig. (Leider nerven dann die LOG-Einträge und dies ist ein WORKAROUND der schlechten Sorte)
> 
> Nun habe ich den Abstand auf 120Minuten genommen und beobachte.
> 
> [@elektron-bbs](https://github.com/elektron-bbs), du hattest Einblick in sämtliche Timer, kommt dir da vielleicht etwas bekannt vor der ein Timer in den sinn wo du das Ganze mit den RSSI Fehler untersucht hattest?

An einem Timer in der Firmware kann es nicht liegen. Da gibt es nur einen, der maximal alle 32 mS den FiFo bei ASK/OOK zurück setzt und die Uptime aktualisiert.
Es muss etwas spezielles bei diesem Protokoll/Einstellungen sein. Bei mir läuft ein SIGNALduino mit Bresser_5_in_1 über mehrere Monate stabil.
Hast du denn mal, wenn der Empfang aussetzt, den SIGNALduino auf verbose 4 gesetzt? Kommt dann noch irgendein Empfang rüber? Ich habe den Verdacht, das es mit Lacrosse Mode 1 zusammenhängt. Das ist die gleiche Datenrate und fast gleiche Frequenz, allerdings unterschiedliche Länge der Nachrichten.

sidey79, 2026-01-08:
Ich wünsche euch auch ein gesundes und erfolgreiches neues Jahr 2026.

Bezüglich dieses Problemes bin ich auch etwas ratlos.

Der uC selbst bleibt nicht hängen, weil er ja weiterhin seine Arbeit verrichtet.

Wenn das Problem vor liegt, dann ist die Kommunikation mit dem cc1101 gestört. In dem Moment müsste man ja mal alle Register usw. bekommen aber ich wüsste nicht, wie man die abfragen kann.

Vielleicht ist einfach nur die SPI Kommunikation gestört? Also ich hatte das Problem mit einem esp32. Seit dem periodischen neu Initialisieren fällt es nicht mehr auf.

sidey79, 2026-01-08:
Der Wert 0xD0 entspricht binär 1101 0000. Die Zählweise der Bits im TI Datenblatt (Seite 72 für PKTSTATUS) ist von 7 (MSB) bis 0 (LSB).

Die Aufschlüsselung für 1101 0000:

Bit | Name | Wert | Bedeutung
-- | -- | -- | --
7 | CRC_OK | 1 | Das vorherige Paket wurde korrekt empfangen (CRC Check ok).
6 | CS | 1 | Carrier Sense. RSSI ist über dem Schwellwert.
5 | PQT_REACHED | 0 | Präambel-Qualität nicht erreicht.
4 | CCA | 1 | Clear Channel Assessment. Der Kanal gilt als "frei".
3 | SFD | 0 | Sync Word nicht gefunden (Kein Paket-Start erkannt).
2 | GDO2 | 0 | (Status des Pins)
1 | - | 0 | (Reserved)
0 | GDO0 | 0 | (Status des Pins)

Zusammengefasst:

CS = 1: Der Chip sagt: "Hier ist ein Signal (oder Rauschen), das lauter ist als mein Schwellwert."
CCA = 1: Der Chip sagt: "Der Kanal ist frei, du darfst senden."

Vermutlich hat der AFC den Empfänger abwandern lassen, ich teste mal mit einem periodischen SIDLE Befehl um neu zu kalibrieren.

sidey79, 2026-01-08:
Ich hab schon mit und ohne AFC getestet... Aktuell wieder mit AFC...aber die Registerwerte bezogen sich ja auf die Daten von @HomeAutoUser

elektron-bbs, 2026-01-08:
Um die Frequenzabweichung zu kontrollieren habe ich mir bei den entsprechenden Sensoren ein Userreading angelegt:

`attr <SENSOR> userReadings freqOffset { InternalVal($name,'sduinoD1_FREQAFC',0) }
`

Die Auswertung sieht dann so aus:

[image]

Ab 10:00 Uhr ist die AFC eingeschaltet.
Da ich die "Wackelei" der Frequenz nicht gut finde, habe ich dann die Frequenz beim SIGNALduino auf einen Mittelwert korrigiert und die AFC ausgeschaltet.

sidey79, 2026-01-08:
Hmm, hab mir jetzt erst mal eingebaut, dass ich mit zwei telnet clients verbinden kann....

So mein AFC ist aus.....
` MN=1;WMBus=0;WMBus_T=0;AFC=0`

Aber die Abweichung wird natürlich trotzdem mitgegeben

`MN;D=40013E3C8D441B2AD120A0780047;R=190;A=-21;`

Bisher hatte ich jetzt leider noch keinen Aussetzer.... Ich hoffe das ändert sich noch

elektron-bbs, 2026-01-19:
> Das kann ich nur erneut bekunden. AFC ist abgeschalten.
> 
> > MS=1;MU=1;MC=1;MN=1;AFC=0;Mred=1

Mhmm, die Kombination bei config kann es nicht geben, gleichzeitig ASK/OOK (MS=1;MU=1;MC=1) und FSK (MN=1;AFC=0)!

HomeAutoUser, 2026-01-19:
Soeben noch einmal "get config" abgerufen und dies ändert sich nicht.

>version: V 4.0.0+20230814 SIGNALduino cc1101 (chip CC1101) - compiled at Mar 13 2024 21:01:41

elektron-bbs, 2026-01-20:
> Was haltet ihr eigentlich von mqtt anstelle telnet?

Ich würde das gern woanders besprechen, vielleicht besser in [Neues Backend auf python Basis](https://github.com/RFD-FHEM/SIGNALDuino/discussions/398)?
Oder verstehe ich deine Frage falsch und du willst dem SIGNALduino MQTT implementieren?
Auf jeden Fall passt es nicht in dieses issue!

sidey79, 2026-01-20:
> Ich würde das gern woanders besprechen, vielleicht besser in [Neues Backend auf python Basis](https://github.com/RFD-FHEM/SIGNALDuino/discussions/398)? Oder verstehe ich deine Frage falsch und du [...]

Ja passt nicht direkt hier her. Das Debuggen der Netzwerk angebundenen Systeme wäre leichter, wenn wir die Debug Meldungen per Netzwerk übermitteln. Da kam mir halt gerade MQTT in den sinn, weil man da einfach ein anderes topic nehmen könnte.
Passt auch nur bedingt zum Backend, aber hat natürlich Einfluss auf alle Backends. Ich starte aber gerne einen neuen Thread dafür.

elektron-bbs, 2026-01-20:
Temperatur halte ich für ausgeschlossen, da ihr sicher den CC1101 nicht solchen extremen Schwankungen aussetzt:
> Since the current calibration values are only
valid for a finite temperature range (typically
±40C) the PLL must be re-calibrated if the lock
indicator does not indicate PLL lock.

Spannungsschwankung könnte schon eher sein.

Eingestellt ist bei mir momentan:
```
0x18  MCSM0    - 0x18 - 0x18 - 00 01 1000
                                  FS_AUTOCAL[1:0] - 01 When going from IDLE to RX or TX (or FSTXON)
0x24  FSCAL2   - 0x2A - 0x0C - 001 0 1010
                                   VCO_CORE_H_EN - Choose high (1) / low (0) VCO                
0x2E  TEST0    - N/A  - 0x0B - 0000 10 1 1
                                       VCO_SEL_CAL_EN - Enable VCO selection calibration stage when 1
```

elektron-bbs, 2026-01-29:
> Ich stelle mir das gerade so vor, wir empfangen ein Datenpaket das liegt im FIFO. Jetzt lesen wir schön aus, aber parallel wird das Nächste Datenpaket empfangen. So wie ich es verstanden habe funktioniert es nicht die Daten komplett auszulesen, wenn parallel noch empfangen wird. Daher sollen wir die Länge im FIFO mehrmals vergleichen und erst dann auslesen, wenn nichts empfangen wird.

Ich weiß nicht, ob der CC1101 noch weiter empfängt, wenn er uns meldet, das das Packet vollständig empfangen wurde.
Wenn wir warten wollen, bis nichts mehr empfangen wird, warten wir u.U. ewig.
Außerdem lesen wir ja das Packet aus und machen anschließend flush  fifo. Bei dir hat offensichtlich das Auslesen nicht funktioniert, denn es lagen noch 14 Byte im FIFO.

Für mich liegt der Fehler irgendwo in diesen beiden subs:

```
uint8_t cc1101::flushrx() {                                // xFSK, Flush the RX FIFO buffer
	if (cmdStrobeTo(CC1101_SIDLE) == false) {
		return false;
	}
	cmdStrobe(CC1101_SNOP);
	cmdStrobe(CC1101_SFRX);
	return true;
}

uint8_t cc1101::cmdStrobeTo(const uint8_t cmd) {    // wait MISO and send command strobe to the CC1101 IC via SPI
	cc1101_Select();                                // select CC1101
	if (waitTo_Miso() == 0) {                       // wait with timeout until MISO goes low
		return false;                               // timeout
	}
	sendSPI(cmd);                                   // send strobe command
	cc1101_Deselect();                              // deselect CC1101
	return true;
}
```

Vielleicht muss nach cmdStrobeTo noch ein delay rein. Ich lese im Internet da so etwas von 100 µs bis 2 mS nach jedem command strobe. Besser wäre allerdings jeweils MARCSTATE zu prüfen.

> PS: Soll ich meine ergänzten Debug meldungen mal pushen? Bisher sind die aber noch nicht ausgelöst worden.

Kannst du gern machen.

sidey79, 2026-02-14:
> Das hat zwar jetzt auch wieder nichts mit dem Issue zu tun, aber trotzdem kurze Frage: Verbindet sich dein ESP32 wieder nach dem WLAN-Ausfall? Bei mir hat das jetzt mehrfach nicht funktioniert. Bei den ESP8266 funktioniert das ohne Probleme.

Ich weiss nicht so genau was Du meinst. Also wenn das WLAN mal kurz weg ist, weil der Router neu gestartet wird, dann verbindet er sich wieder. Vom Prinzip merke ich nichts, weil ja alles funktioniert.
Was genau geschah, als er keine WLAN Verbindung mehr hergestellt hat, kann ich nur raten, denn ich konnte nicht auf ihn zugreifen.

HomeAutoUser, 2026-04-13:
Hallo @elektron-bbs, ich habe nun mal einarmig mich der Sache gewidmet.

> [@HomeAutoUser](https://github.com/HomeAutoUser) Hast du denn eigentlich schon mal diese Firmware probiert? https://github.com/RFD-FHEM/SIGNALDuino/tree/master_CC1101-PLL

Diese FW habe ich nun auf die Empfänger geflasht. Jetzt muss ich mal schauen ob der Empfang erfolgt und wie.

elektron-bbs, 2026-10-29:
Es scheint so, als ob dieses Problem seit ca. 6 Monaten nicht wieder aufgetreten ist???

HomeAutoUser, 2026-10-29:
Kaum bekommst du mit ;) das ich etwas mehr Zeit habe, erfolgt die Nachfrage *smile*
Das Problem besteht weiterhin, da ich immer noch die GatewaySW am laufen habe. Ich werde extra für Euch am WE diese auf die SignalduinoFW umstellen um das Problem zu provozieren.

sidey79, 2026-10-29:
Also bei mir ist es nicht mehr aufgetreten..

HomeAutoUser, 2026-11-02:
Ich habe nun begonnen einen weiteren ESP (solo) in den Empfang der SD_WS125 einzubinden.
Mal schauen wie dieser sich verhält. Der nächste Step ist, die 3 uC wieder zu aktivieren um einen Ausfall zu provozieren.

sidey79, 2026-11-02:
Ich habe einige mögliche Hänger im Zusammenhang mit Sendekommandos identifiziert.
In einem Fall wird der Empfänger nicht mehr eingeschaltet. 

Das sind alles ziemlich seltene Fälle, aber sie können auftreten.

Nutzt Du Sendekommandos regelmäßig?

HomeAutoUser, 2026-11-02:
Ich nutze nur Receive mit dem Protokoll. Es kommt aber bei 8 Sensoren sehr häufig und teilweise "zeitgleich".
Sobald ich die Umgebung hergestellt habe, würde ich mit dem Branch von @elektron-bbs noch einmal versuchen das STOP Verhalten zu vergleichen.